microsoft/AirSim #2694: Xbox 306 controller on OSX

galgyeony, 2020-05-15:
By referring to https://github.com/microsoft/AirSim/blob/master/docs/build_linux.md,
I successfully installed the UE4(4.24.3) and AirSim(the newest as of today) on OSX(Catalina 10.15.4).
The blocks project was also built well.
Furthermore, I installed the Xbox 306 controller driver(1.0.0-alpha.5) and checked it works well in System Preferences in OSX.

However, the multirotor in the blocks project cannot be controlled manually via the Xbox 306 controller (while the car is controlled well with keyboard).
Is there anyone who knows the solution?

Thanks in advance.

rajat2004, 2020-05-16:
AFAIK, I don't think people have tried using RC on OSX. Till recently, to run AirSim on OSX, some manual changes in the setup scripts and maybe code was required. Only with the recent upgrades to Clang 8, etc has it been tested to be working properly.
If you would like to use RC on OSX, you'll probably have to dig a bit through the code and figure out why things aren't working. [SimJoystick](https://github.com/microsoft/AirSim/tree/master/Unreal/Plugins/AirSim/Source/SimJoyStick) has the files which interface with them, in that, there are separate pieces of code for Windows and other OSes, you could try to add debug code to see any errors, etc.

A place to start could be https://github.com/microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/SimJoyStick/SimJoyStick.cpp#L278 where it's opening the device, the path could be different for OSX

galgyeony, 2020-05-18:
@rajat2004 Thank you so much for your advise. I am not accustomed to program in OSX but anyhow tried several ways (e.g., /dev/tty instead of /dev/input/js) but failed. :(

rajat2004, 2020-05-18:
I myself have no experience working with OSX code, let alone serail devices so can't help be of much help. Another way to start could be to figure out differences between OSX and Linux when working with serial devices, taking input, etc.
Note that what I had pointed out was just one thing, I could be wrong also that it's different between Linux and OSX. Or more changes could be required.