microsoft/AirSim #2695: simGetObjectPose() sometimes returns NaN

MSBGit, 2020-05-15:
`simGetObjectPose()` sometimes returns NaN values due to a bug in UE4 API that is used to retrieve Actor information. The common way to deal with this in client code is to retry the call to `simGetObjectPose()` a number of times until valid data is returned or until the caller decides that it has made too many calls or waited too long for valid data. It would be great if AirSim would do the retries under the hood and not require clients of AirSim to do this. Then when/if the UE4 bug is fixed, client code doesn't need to change.

rajat2004, 2020-05-16:
This could also be due to the Actor not being found
https://github.com/microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/WorldSimApi.cpp#L75
Is there any bug report or forum post regarding this?

As to checking and retrying, [ContainsNaN](https://docs.unrealengine.com/en-US/API/Runtime/Core/Math/FVector/ContainsNaN/index.html) could be used for this I guess. No of retries and timeout will have to decided

MSBGit, 2020-05-20:
This happens every so often, particularly when I am calling `simGetObjectPose()` at a 1Hz rate. I always call it for the same actor ID or tag.

> Is there any bug report or forum post regarding this?

Do you mean a bug report or forum post in an Unreal bug system or forum? I haven't found any but I didn't do an exhaustive search.

Here is one example of how I deal with this in one of my Python modules:

rajat2004, 2020-05-20:
That does look quite painful. I've added a utility function in https://github.com/microsoft/AirSim/pull/2683 which will atleast reduce the amount of code required.
I agree, this could be checked inside AirSim itself, and called again if the Pose is Nan. However, this can be done only a few no. of times, and even then I think a check from the client side will be required just to be sure

MSBGit, 2020-05-20:
> That does look quite painful. I've added a utility function in #2683 which will at least reduce the amount of code required.

Thanks. I'll see if I can find the underlying Unreal issue that causes this.

jonyMarino, 2020-10-16:
@MSBGit I'm not sure if was fixed in UE 4.24.3 but I'm not getting NaN never. Can you confirm?

jonyMarino, 2020-11-04:
This issue got fixed in PR 2898, which added an asset map and scene object map

closed 2020-11-04 by jonyMarino