microsoft/AirSim #2696: How do I disable animals in Neighborhood environment?

GershomA, 2020-05-16:
I'm trying to use airsim's Neighborhood scene as an RL environment.
One of my "fail conditions" is collisions.
Animals trigger this by running into the vehicle, or standing on the spawn point, clipping into the car and launching it off to space the second it spawns 🤣 
Is there a way to configure the neighborhood environment to disable animal spawns? I haven't found one in the docs

rajat2004, 2020-05-17:
Lol :laughing: 
There are PRs to add such an API, see https://github.com/microsoft/AirSim/pull/2651, which has lots of others as well, https://github.com/microsoft/AirSim/pull/2561 which was closed in favour of the former
However there's no such functionality in the current binaries at least

A lengthy workaround could be to use `simGetCollisionInfo` and check the object name, and if it matches with the animals, then ignore that. Not a pretty one, but might work

GershomA, 2020-05-17:
Thanks, I'll use that for now and just hard reset on the lunar launches 👍

closed 2020-05-17 by GershomA