microsoft/AirSim #2697: Windows install issue

EdwinMeriaux, 2020-05-17:
Hi,

So I was trying to install airsim for Windows and I ran into an issue when I tried to launch the blocks.uproject for the first time. I get this error:

the following modulces are missign or built with a different engine version:

blocks
Airsim
would you like to rebuild them now?

I say yes

then I proceed to get:
Blocks could not be compiled. Try rebuilding from source manually.

I know in the FAQ (https://github.com/Microsoft/AirSim/blob/123bbd6d6416bbe98e775101ba8ebcb5ccbf8501/docs/build_linux.md#im-getting-error--could-not-be-compiled-try-rebuilding-from-source-manually)
if talks about this but it only referes for linux no? 
So how would I solve this?

rajat2004, 2020-05-17:
Windows would also give a similar error message, but yeah, the specific error described in the Readme only applies to Linux. 
Since you're running for the first time, did you complete the steps given in https://github.com/microsoft/AirSim/blob/master/docs/unreal_blocks.md#windows? `update_from_git.bat` is imp since it copies the plugin files and generates the VS solution file. Directly running the .uproject will fail as there's no plugin present right now

EdwinMeriaux, 2020-05-17:
So I able to create the sln file but I had an issue opening it in visual studio 

the issue with the sln file was it says:
the following files were specified on the command linke:
c:\Windows\System32\AirSim\Unreal|Environments\Blocks\Vlocks.sln
These files could not be found and will not be loaded

I presume this is an issue why I get the issue launching the block.uproject?
how would I fix this?

rajat2004, 2020-05-17:
Try running `clean.bat` & `update_from_git.bat` again from the Blocks directory
Any errors while running update_from_git.bat?

EdwinMeriaux, 2020-05-17:
there were no errors running update_from_git.bat and the clean.bat before that.
but I still cannot open the sln file in visual studio

rajat2004, 2020-05-17:
Hmm, your error message is saying that the .sln file doesn't exist
Do you have VS 2017 & VS 2019 both installed?

EdwinMeriaux, 2020-05-17:
Yes I have both. I initially installed VS 2019 but then I was having issues using that.
I found tips online saying I can use VS 2017.

rajat2004, 2020-05-17:
Current AirSim master requires VS 2019, see [this](https://github.com/microsoft/AirSim/blob/master/docs/build_windows.md#unreal-still-uses-vs2015-or-im-getting-some-link-error) for details on changing the VS version for UE, if that doesn't work, then try the new tips [here](https://github.com/microsoft/AirSim/pull/2658/files#diff-c0b760de2909db03427468d7966967b0)

EdwinMeriaux, 2020-05-17:
So when I try to use VS 2019 this is the issue I run into:

CMake Error: The source directory "C:/Windows/System32/AirSim/external/rpclib/rpclib-2.2.1" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
Error: C:/Windows/System32/AirSim/external/rpclib/rpclib-2.2.1/build is not a directory
Error: C:/Windows/System32/AirSim/external/rpclib/rpclib-2.2.1/build is not a directory

rajat2004, 2020-05-17:
Well, then it seems `build.cmd` didn't work properly since that downloads rpclib.
I would suggest you run `clean.cmd` from the root directory and then do all the steps as mentioned in https://github.com/microsoft/AirSim/blob/master/docs/build_windows.md
Make sure to check for any error messages in the scripts

EdwinMeriaux, 2020-05-17:
when I try to run cmd.clean:

C:\Windows\System32\AirSim>clean.cmd

C:\Windows\System32\AirSim>rd /s/q AirLib\lib
The system cannot find the file specified.

C:\Windows\System32\AirSim>rd /s/q AirLib\deps\MavLinkCom
The system cannot find the path specified.

C:\Windows\System32\AirSim>rd /s/q AirLib\deps\rpclib
The system cannot find the path specified.

C:\Windows\System32\AirSim>rd /s/q external\rpclib\build
The system cannot find the file specified.

C:\Windows\System32\AirSim>msbuild /p:Platform=x64 /p:Configuration=Debug AirSim.sln /t:Clean
Microsoft (R) Build Engine version 16.5.1+4616136f8 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

MSBUILD : error MSB1009: Project file does not exist.
Switch: AirSim.sln

C:\Windows\System32\AirSim>if ERRORLEVEL 1 goto :buildfailed

C:\Windows\System32\AirSim>echo #### Build failed
#### Build failed

C:\Windows\System32\AirSim>goto :eof

rajat2004, 2020-05-17:
Then run `build.cmd` directly and then all the following steps

EdwinMeriaux, 2020-05-17:
Yes, but I still get the same issue when running build.cmd:

C:\Windows\System32\AirSim>build.cmd
Found cmake version: 3.16.19112601-MSVC_2
Starting cmake to build rpclib...
CMake Error: The source directory "C:/Windows/System32/AirSim/external/rpclib/rpclib-2.2.1" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
Error: C:/Windows/System32/AirSim/external/rpclib/rpclib-2.2.1/build is not a directory
Error: C:/Windows/System32/AirSim/external/rpclib/rpclib-2.2.1/build is not a directory

#### Build failed - see messages above.

C:\Windows\System32\AirSim>

rajat2004, 2020-05-17:
Okay, that's a pretty strange issue happening. `build.cmd` should create the folder and download rpclib if it doesn't exist. Is your internet connection working properly?
Easiest way I can think of right now is to delete the entire AirSim repo folder, clone again and do the steps again

EdwinMeriaux, 2020-05-17:
Oh I guess that solved most of the issues.

There is one remaining one though. Now I can open the sln file in visual studio but I see this
[image]

rajat2004, 2020-05-18:
This _might_ be happening due to AirSim being installed in C drive but not sure. Try running VS in admin mode. Else googe to see what could cause this problem

EdwinMeriaux, 2020-05-18:
Yeah so that was the remaining issue. Why did I need to be in admin mode in VS?

EdwinMeriaux, 2020-05-18:
Thank you so much for all your help!!!

rajat2004, 2020-05-18:
I think it's better to not install stuff in the C drive unless required, kind of like directly installing stuff in root directory or always being the superuser in Linux, which isn't a good thing. Will update the PR to add a note regarding this in the docs
If the probem is solved, could you close the issue?

closed 2020-05-18 by EdwinMeriaux