microsoft/AirSim #2698: Controlling two drones using keyboard keys

AngelTang190, 2020-05-18:
Is it possible to control two drones in AirSim using keyboard keys? 
I tried to change the settings.json file and now I have two drones in my project now. However, I do not know how to control them using keyboard keys in AirSim. I would like to create the codes using Python. Can anyone help please?

rajat2004, 2020-05-18:
Currently, you can't control drones using keyboard. You could do something like take keyboard input in Python and call some APIs depending on action desired
Also for mutliple drones, you'll have to set the name of the drone you want to send the command to using `vehicle_name` arg in the APIs

AngelTang190, 2020-05-18:
> Currently, you can't control drones using keyboard. You could do something like take keyboard input in Python and call some APIs depending on action desired
> Also for mutliple drones, you'll have to set the name of the drone you want to send the command to using `vehicle_name` arg in the APIs

Because currently I am working on a research which requires me to fly at least two drones or more in AirSim, and record their data if the drones collide. I was thinking that if I can control the drones, then it will be easier for me to produce collisions among the drones, However, since now I cannot control the drones using keyboard, is there any other way I can create collisions among drones?

rajat2004, 2020-05-18:
There are lot's of APIs available for controlling the movement of drones, see https://github.com/microsoft/AirSim/blob/master/docs/apis.md. For the most up-to-date list of APIs, see https://github.com/microsoft/AirSim/blob/master/PythonClient/airsim/client.py, specifically the `moveBy*` ones. There's a PR open to convert the docstrings to a reference doc containing all the APIs - #2658 

I think controlling them through APIs will be the best way, it's very difficult to to control them using keyboard due to the limited options, and it's not repeatable. On the other hand, you can easily run the same script hundreds of times

AngelTang190, 2020-05-19:
> There are lot's of APIs available for controlling the movement of drones, see https://github.com/microsoft/AirSim/blob/master/docs/apis.md. For the most up-to-date list of APIs, see [...]

Thank you! 

Can you briefly describe what does the codes in https://github.com/microsoft/AirSim/blob/master/PythonClient/airsim/client.py does?

Can I know how to I set the vehicle names? Are there any tutorials or websites available? Besides, can I run the same file (e.g. hello_drone.py) for all drones simultaneously?

rajat2004, 2020-05-19:
`client.py` is just a wrapper file which calls the actual functions defined in C++. My main point of mentioning it was that to have a look at it for all the APIs available since many of the new ones aren't documented properly elsewhere

Re multiple vehicles, as linked in the API doc itself, see https://github.com/microsoft/AirSim/blob/master/docs/multi_vehicle.md for details

AngelTang190, 2020-05-19:
Ok , thank you! 

By thee way, when I tried to run multi_agent_drone.py, my AirSim crashed. Should I open a new issue?

rajat2004, 2020-05-19:
Script's working properly for me. Are you running the latest master, or any release? If using a release, make sure you're running the latest ones

AngelTang190, 2020-05-19:
import airsim
import cv2
import numpy as np
import os
import pprint
import setup_path 
import tempfile

# Use below in settings.json with Blocks environment
"""
{
	"SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
	"SettingsVersion": 1.2,
	"SimMode": "Multirotor",
	"ClockSpeed": 1,
	
	"Vehicles": {
		"Drone1": {
		  "VehicleType": "SimpleFlight",
		  "X": 4, "Y": 0, "Z": -2
		},
		"Drone2": {
		  "VehicleType": "SimpleFlight",
		  "X": 8, "Y": 0, "Z": -2
		}
    }
}
"""

# connect to the AirSim simulator
client = airsim.MultirotorClient()
client.confirmConnection()
client.enableApiControl(True, "Drone1")
client.enableApiControl(True, "Drone2")
client.armDisarm(True, "Drone1")
client.armDisarm(True, "Drone2")

airsim.wait_key('Press any key to takeoff')
f1 = client.takeoffAsync(vehicle_name="Drone1")
f2 = client.takeoffAsync(vehicle_name="Drone2")
f1.join()
f2.join()

state1 = client.getMultirotorState(vehicle_name="Drone1")
s = pprint.pformat(state1)
print("state: %s" % s)
state2 = client.getMultirotorState(vehicle_name="Drone2")
s = pprint.pformat(state2)
print("state: %s" % s)

airsim.wait_key('Press any key to move vehicles')
f1 = client.moveToPositionAsync(-5, 5, -10, 5, vehicle_name="Drone1")
f2 = client.moveToPositionAsync(5, -5, -10, 5, vehicle_name="Drone2")
f1.join()
f2.join()

airsim.wait_key('Press any key to take images')
# get camera images from the car
responses1 = client.simGetImages([
    airsim.ImageRequest("0", airsim.ImageType.DepthVis),  #depth visualization image
    airsim.ImageRequest("1", airsim.ImageType.Scene, False, False)], vehicle_name="Drone1")  #scene vision image in uncompressed RGB array
print('Drone1: Retrieved images: %d' % len(responses1))
responses2 = client.simGetImages([
    airsim.ImageRequest("0", airsim.ImageType.DepthVis),  #depth visualization image
    airsim.ImageRequest("1", airsim.ImageType.Scene, False, False)], vehicle_name="Drone2")  #scene vision image in uncompressed RGB array
print('Drone2: Retrieved images: %d' % len(responses2))

tmp_dir = os.path.join(tempfile.gettempdir(), "airsim_drone")
print ("Saving images to %s" % tmp_dir)
try:
    os.makedirs(tmp_dir)
except OSError:
    if not os.path.isdir(tmp_dir):
        raise

for idx, response in enumerate(responses1 + responses2):

    filename = os.path.join(tmp_dir, str(idx))

    if response.pixels_as_float:
        print("Type %d, size %d" % (response.image_type, len(response.image_data_float)))
        airsim.write_pfm(os.path.normpath(filename + '.pfm'), airsim.get_pfm_array(response))
    elif response.compress: #png format
        print("Type %d, size %d" % (response.image_type, len(response.image_data_uint8)))
        airsim.write_file(os.path.normpath(filename + '.png'), response.image_data_uint8)
    else: #uncompressed array
        print("Type %d, size %d" % (response.image_type, len(response.image_data_uint8)))
        img1d = np.fromstring(response.image_data_uint8, dtype=np.uint8) #get numpy array
        img_rgb = img1d.reshape(response.height, response.width, 3) #reshape array to 3 channel image array H X W X 3
        cv2.imwrite(os.path.normpath(filename + '.png'), img_rgb) # write to png

airsim.wait_key('Press any key to reset to original state')

client.armDisarm(False, "Drone1")
client.armDisarm(False, "Drone2")
client.reset()

# that's enough fun for now. let's quit cleanly
client.enableApiControl(False, "Drone1")
client.enableApiControl(False, "Drone2")

This is the multi_agent_drone.py that I have. It is the latest version I believe.

rajat2004, 2020-05-19:
Not the script, I mean did you build AirSim from source or downloaded a release binary and running that?

AngelTang190, 2020-05-19:
Build AirSim
Install Visual Studio 2019.
Make sure to select Desktop Development with C++ and Windows 10 SDK 10.0.18362 (should be selected by default) while installing VS 2019.
Start Developer Command Prompt for VS 2019.
Clone the repo: git clone https://github.com/Microsoft/AirSim.git, and go the AirSim directory by cd AirSim.
Run build.cmd from the command line. This will create ready to use plugin bits in the Unreal\Plugins folder that can be dropped into any Unreal project.

I followed these commands here to build. So I believe it is from the source.

rajat2004, 2020-05-19:
It's working in the latest master for me. The same settings as given in the script? What's the output when running the script?

AngelTang190, 2020-05-19:
Yes the settings are the same. The output you mean is the output from AirSim or from the code?

rajat2004, 2020-05-19:
Both would be useful, but main is the output from script to see where it failed. Just to confirm, no changes in the script as well right?

AngelTang190, 2020-05-19:
No changes made in the script.

How can I display the output from the script to you?

rajat2004, 2020-05-19:
You would be running the script in Anaconda or Python terminal right? That would have the output, just copy-paste it

AngelTang190, 2020-05-19:
(base) C:\Users\ASUS>cd\

(base) C:\>cd AirSim\PythonClient\multirotor\

(base) C:\AirSim\PythonClient\multirotor>python multi_agent_drone.py
Connected!
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

Press any key to takeoff

This is the code from Anaconda, however the Unreal Engine just crashed

AngelTang190, 2020-05-19:
LoginId:5690cc664b1c5fda56b6afa9e10052bb
EpicAccountId:935f49a690214f7e949826002d647237

Unhandled Exception: 0xe06d7363

msvcp140
UE4Editor_AirSim!<lambda_69f58446dda213d70a34b34d463075c6>::operator()() [C:\Users\ASUS\Documents\Unreal Projects\LandscapeMountains\Plugins\AirSim\Source\RenderRequest.cpp:72]
UE4Editor_AirSim!TBaseFunctorDelegateInstance<void __cdecl(void),<lambda_69f58446dda213d70a34b34d463075c6> >::ExecuteIfSafe() [C:\Program Files\Epic Games\UE_4.24\Engine\Source\Runtime\Core\Public\Delegates\DelegateInstancesImpl.h:1002]
UE4Editor_Engine
UE4Editor_Engine
UE4Editor_Engine
UE4Editor_UnrealEd
UE4Editor_UnrealEd
UE4Editor
UE4Editor
UE4Editor
UE4Editor
UE4Editor
kernel32
ntdll

This is the crash report summary from Unreal Engine

rajat2004, 2020-05-19:
Seems very strange, and other scripts such as hello_drone.py are working fine?

AngelTang190, 2020-05-19:
hello_drone.py is working fine

rajat2004, 2020-05-19:
Well, I'm unable to reproduce on Linux at least. You could create a new issue so that if anyone has any idea, they can have a look.
Also, try running Blocks in debugging mode from Visual Studio to get stacktrace, could also try putting breakpoints or just printfs to see where things could be going wrong

AngelTang190, 2020-05-19:
Ok, I will open a new issue soon.

So how should I go about running Blovks in debugging mode from VS?

AngelTang190, 2020-05-21:
> Well, I'm unable to reproduce on Linux at least. You could create a new issue so that if anyone has any idea, they can have a look.
> Also, try running Blocks in debugging mode from Visual Studio to [...]

I tried to run it in VS and there is this error in line 976 in "functional" of Blocks.sln
Exception thrown: read access violation.
_Impl-> was 0x200000001.

[image]

rajat2004, 2020-05-21:
I haven't yet had time to look into this, but you could try out one of the release binaries for Windows which can be downloaded from [here](https://github.com/microsoft/AirSim/releases). If the problem is occurring in master, then it will probably happen in the release as well, but just to confirm

AngelTang190, 2020-05-21:
Do I have to uninstall the AirSim that I have now?

rajat2004, 2020-05-21:
No, the binaries are stand-alone and don't need anything else. It's actually recommended that people who are using AirSim for the first time try out some of the binaries just to play around, since no setup is required. Binaries are generally faster than running inside Editor.
Also see https://github.com/microsoft/AirSim/blob/master/docs/use_precompiled.md for some command-line options to increase the performance

AngelTang190, 2020-05-21:
Do I have to clone the master commit ID in the link from you just now? Because when I cloned it it says that AirSim is not empty.

rajat2004, 2020-05-21:
No, there are assets present in the release, which are zip files containing the binaries, download any of them, say Blocks, unzip and then run the binary

AngelTang190, 2020-05-21:
How can I download Blocks specifically instead of downloading the whole project?

rajat2004, 2020-05-21:
[image]
.

AngelTang190, 2020-05-21:
Sorry but where can I find assets? I only saw commits , branches and stuff like that.

rajat2004, 2020-05-21:
[image]
There's a button on the left of Assets

AngelTang190, 2020-05-21:
Is this only can be run on Linux and not Windows? And there is no uproject for me to generate solution files and run it

rajat2004, 2020-05-21:
If Windows releases are there, then it should be obvious that it works on Windows. They are executable files, not project files, so solution files can't be generated. Just double click on the .exe file to run

AngelTang190, 2020-05-21:
I only see the shell script and I don't see .exe file

AngelTang190, 2020-05-21:
The application runs perfectly fine for multi_agent_drone.py

rajat2004, 2020-05-21:
Well, there haven't been any changes in master which would affect this. Maybe some other problem in UE Editor. Since there's already an issue open to track the problem, this can be closed

AngelTang190, 2020-05-21:
OK, thank you!

closed 2020-05-21 by AngelTang190