microsoft/AirSim #2699: Unable to use SITL with Airsim on Unreal Engine

clev0910, 2020-05-19:
I have setup AirSim with the landscape mountains environment and have also set up SITL using Cygwin. However, when I attempt to run SITL in order to move the quadcopter around on unreal engine, while I do get MAVlink, the map and the console to open, any commands that I type into the MAVlink have no effect on the quadcopter; it does not arm nor takeoff or do anything after that. I do receive a message as I attempt to start SITL, an image of which I've attached below. I've also attached the settings I've used in the .json settings folder for the Unreal Project. Is there any change that needs to be made to the settings/SITL set up?
[
[image]

[image]

](url)

agentfeux, 2020-05-21:
I don't know if its gonna help but i personally do on the console: `sim_vehicle.py -v ArduCopter -f airsim-copter`

Then on another tab: `./Engine/Binaries/Linux/UE4Editor ~/AirSim/Unreal/Environments/Blocks/Blocks.uproject`

I use QgroundControl for the map and stuff.

rajat2004, 2020-05-22:
Try using WSL instead of Cygwin for running ArduPilot, both WSL 1 & 2 have been tested to be working with AP+Airsim
Cygwin has some strange networking issues, you could try using `0.0.0.0` for IP addresses as instead of `127.0.0.1` as described [here](https://ardupilot.org/dev/docs/sitl-with-airsim.html#run-on-different-machines)

jonyMarino, 2021-04-22:
Closed due to inactivity. Please, feel free to ask for it to be reopened if you wish to continue posting.

closed 2021-04-22 by jonyMarino

AhmedSaleh627, 2024-10-27:
Iam facing the same issue