microsoft/AirSim #2701: AirSim crashes when running multi_agent_drone.py

AngelTang190, 2020-05-19:
# Read This First

## If you are reporting a bug
* Make sure to write **all reproduction steps**
* Include full error message in text form
* Search issues for error message before filing issue
* Attach screenshot if applicable
* Include code to run if applicable

## If you have question
* Add clear and concise title
* Add OS, AirSim version, Python version, Unreal version if applicable
* Include context on what you are trying to achieve
* Include details of what you already did to find answers

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------

I opened AirSim and hit the Play button. After that, I used anaconda3 and typed in the following:

(base) C:\Users\ASUS>cd\

(base) C:>cd AirSim\PythonClient\multirotor\

(base) C:\AirSim\PythonClient\multirotor>python multi_agent_drone.py

And I hit enter.

Then anaconda gave me this:
Connected!
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

Press any key to takeoff

I pressed enter. After that Unreal Engine closes abruptly and gave me this summary :

LoginId:5690cc664b1c5fda56b6afa9e10052bb
EpicAccountId:935f49a690214f7e949826002d647237

Unhandled Exception: 0xe06d7363

msvcp140
UE4Editor_AirSim!<lambda_69f58446dda213d70a34b34d463075c6>::operator()() [C:\Users\ASUS\Documents\Unreal Projects\LandscapeMountains\Plugins\AirSim\Source\RenderRequest.cpp:72]
UE4Editor_AirSim!TBaseFunctorDelegateInstance<void __cdecl(void),<lambda_69f58446dda213d70a34b34d463075c6> >::ExecuteIfSafe() [C:\Program Files\Epic Games\UE_4.24\Engine\Source\Runtime\Core\Public\Delegates\DelegateInstancesImpl.h:1002]
UE4Editor_Engine
UE4Editor_Engine
UE4Editor_Engine
UE4Editor_UnrealEd
UE4Editor_UnrealEd
UE4Editor
UE4Editor
UE4Editor
UE4Editor
UE4Editor
kernel32
ntdll

Can I know what is wrong and how to fix this?

rajat2004, 2020-05-21:
The problem is not reproducible on Linux with master as well as the latest binary, needs investigation on Windows

AngelTang190, 2020-05-21:
How can I fix my Unreal Engine so that it does not crash?

rajat2004, 2020-05-21:
I have no idea what's the problem here, Windows releases are also working fine. No changes have been made to master which could cause this effect. Only difference I can think of is system resources, but that's very unlikely

AngelTang190, 2020-05-21:
I tried to run multi_agent_drone.py in my Blocks Environment and Landscape Mountains environment. Both of them can only run hello_drone.py but not multi_agent_drone.py

AngelTang190, 2020-05-21:
Can I record the coordinates of both of the drones when I run multi_drone_agent.py?

closed 2020-05-28 by AngelTang190