microsoft/AirSim #2708: Problem building AirSim on Arch Linux

itsMatoosh, 2020-05-21:
I tried building the AirSim library for Unity using the following guide - https://microsoft.github.io/AirSim/Unity/. I'm running Manjaro Linux, an Arch distro. I tried running the setup.sh and build.sh scripts as mentioned in the guide, however, I get errors because the apt-get package manager which the scripts are calling is absent on arch. Is there a way to build AirSim on Arch? If so, how do I go about this?

rajat2004, 2020-05-21:
You could install clang-8 and associated packages manually, and then comment out the part installing clang

itsMatoosh, 2020-05-21:
@rajat2004 clang-8 is not available through pacman. Do you know how I could install it on arch?

rajat2004, 2020-05-21:
That you would have to figure out by googling, I haven't used Arch so can't say
LLVM website could have binaries which can be downloaded, or manual build from source code is also possible as an option

itsMatoosh, 2020-05-21:
I managed to build it successfully inside of the Anaconda environment. For anyone trying to build on arch, install Anaconda and then use the following commands.
`conda create --name airsim`
`conda activate airsim`
`conda install -c conda-forge boost`
`conda install -c conda-forge clang=8.0.1`
`conda install -c conda-forge libcxx=8.0.1`
`./setup.sh`
`./build.sh`

closed 2020-05-21 by itsMatoosh

rajat2004, 2020-05-21:
Well, that's pretty interesting, didn't know that was possible