microsoft/AirSim #2717: navigation with point cloud from depth map

EdwinMeriaux, 2020-05-25:
So using 
- https://github.com/microsoft/AirSim/blob/master/PythonClient/multirotor/point_cloud.py
- https://github.com/Microsoft/AirSim/wiki/Point-Clouds
- https://github.com/Microsoft/AirSim/blob/master/docs/image_apis.md#what-does-pixel-values-mean-in-different-image-types

I understand how to take images in airsim and make a depth map and then convert that to a point cloud. But is there a way to get a drone or car in airsim to navigat using the point cloud?
I just haven't been able to figure out an answer to this.

rajat2004, 2020-05-25:
PointCloud is like output from sensor, just like an image. Extra work is needed to be done on the data to make it useful, in case of pointcloud, it could be passed onto some ROS controllers such as MoveIt which could take PCL as input and try to avoid obstacles.
Just some idea off the top of my head, you'll need to search and use it based on your requirement

EdwinMeriaux, 2020-05-25:
> PointCloud is like output from sensor, just like an image. Extra work is needed to be done on the data to make it useful, in case of pointcloud, it could be passed onto some ROS controllers such as [...]

ok Awesome!

Thank you for the tip!

One more question from the depth image that can be made with airsim. How do I save the point cloud or at least the depth image to my computer?

closed 2020-05-28 by EdwinMeriaux