microsoft/AirSim #2721: Linker Error on Catalina

ahorvat, 2020-05-25:
When compiling on MacOS 10.15.4, with XCode 11.5 I receive the following error.

```
Undefined symbols for architecture x86_64:
  "___darwin_check_fd_set_overflow", referenced from:
      mavlinkcom_impl::UdpSocketImpl::pollin(unsigned int) in UdpSocketImpl.cpp.o
      mavlinkcom_impl::UdpSocketImpl::pollout(unsigned int) in UdpSocketImpl.cpp.o
ld: symbol(s) not found for architecture x86_64
clang-8: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [AirsimWrapper.bundle/Contents/MacOS/AirsimWrapper] Error 1
make[1]: *** [CMakeFiles/AirsimWrapper.dir/all] Error 2
make: *** [all] Error 2
+ '[' '!' -d ../UnityDemo/Assets/Plugins ']'
+ mkdir ../UnityDemo/Assets/Plugins
++ uname
+ '[' Darwin == Darwin ']'
+ cp -r AirsimWrapper.bundle ../UnityDemo/Assets/Plugins
+ cd ..
+ '[' -f AirLibWrapper/AirsimWrapper/rpclib.pc.in ']'
+ rm AirLibWrapper/AirsimWrapper/rpclib.pc.in
```

rajat2004, 2020-05-25:
This is a known bug in XCode, and is being discussed on #2559, it'll be better to keep it on the same topic

ahorvat, 2020-05-25:
> This is a known bug in XCode, and is being discussed on #2559, it'll be better to keep it on the same topic

Thanks, I missed that.

closed 2020-05-25 by ahorvat