microsoft/AirSim #2723: Visualizing sensors data in Client side (Unreal)

springyuniverse, 2020-05-25:
I am trying to visualise sensors data (Distance and Lidar sensors data ) in unreal engine, what the right way to approach this

rajat2004, 2020-05-27:
Lidar has a field `DrawDebugPoints` which when set to `true` will visualize the points. Note that enabling that in the current release can slowdown and cause a crash due to excessive memory usage. This issue has been fixed in master

springyuniverse, 2020-05-27:
Thanks, what about the distance sensor?

rajat2004, 2020-05-27:
Don't think that functionality is there in Distance sensor, but a possibly quick way could be to use the Plotting APIs
Some transformation would probably be required, else could try to modify the codebase, should be somewhat similar to Lidar sensor visualization, except less complex https://github.com/microsoft/AirSim/blob/master/PythonClient/airsim/client.py#L559

springyuniverse, 2020-05-27:
Should I also edit the codebase for also speed and height  of the drone?

springyuniverse, 2020-05-27:
[image]
I mean logging the sensor data in the area as shown in the image

rajat2004, 2020-05-27:
[`simPrintLogMessage`](https://github.com/microsoft/AirSim/blob/master/docs/apis.md#common-apis) can be used to print stuff there, else could modify in codebase also
See [Multirotor APIs](https://github.com/microsoft/AirSim/blob/master/docs/apis.md#getmultirotorstate) on how to get various quantities related to drone state

PR #2807 (2020-07-24): Distance sensor Upgrades
1. Add setting for drawing debug point with Distance Sensor
2. Expose Distance sensor settings, also move sensor above Car vehicle by 1 meter by default 
3. Fix `DistanceSensorData` struct in `types.py` (Should be float, not Quaternion)
4. Update docs
5. Use Custom sensor orientation in Mavlink Distance Sensor message (Needs testing with PX4)
6. ROS wrapper change
7. Example script for Distance sensor with 2 cars

Default sensor-
[image]

Downward-facing (altitude measurement)-
[image]

2 Cars with distance sensor
[image]

Closes #2723, closes #2806
Related issues - #2 [... cut]
closed 2020-07-24 by madratman