microsoft/AirSim #2724: depth image format

EdwinMeriaux, 2020-05-26:
With:
https://github.com/Microsoft/AirSim/blob/master/docs/image_apis.md#what-does-pixel-values-mean-in-different-image-types

I understand that a depth image can be made but I am confused how to store the image and what format the image will be stored as.

EdwinMeriaux, 2020-05-27:
Ok I now know the file format for this. But where is it being stored, for example, in the Hello_drone example?

rajat2004, 2020-05-27:
`hello_drone.py` stores the files in a temporary directory, in Linux, it's generally `/tmp` , on Win, probably `C:\temp` or something, this will be printed in the console when running the script
https://github.com/microsoft/AirSim/blob/master/PythonClient/multirotor/hello_drone.py#L60

EdwinMeriaux, 2020-05-27:
Oh ok thank you! I found it.

closed 2020-05-27 by EdwinMeriaux