microsoft/AirSim #2725: Lidar is not working for certain objects in my custom unreal environment

chickenbestlover, 2020-05-26:
Hi, I'm working on an Airsim project with a custom unreal environment purchased from a market store.

Following the `docs/lidar.md` documentation, I added a lidar sensor to the vehicle.
Also, I set `"DrawDebugPoints": true` to check if the lidar sensor works well.
As shown in the figure below, the lidar sensor cannot sense certain objects (tunnel wall in this figure).
[image]

In order to check if this problem appears only in my environment, I tested the lidar sensor in the City environment. The results are shown in the figure below. Even in this environment, the lidar sensor does not sense some objects (trees).
 [image]

I speculated that this could be solved by changing the properties of the non-sensible objects in the unreal editor. But as a newbie, I couldn't find the difference in the settings between the sensible and non-sensible objects in the editor.

 I'll be grateful for any help or recommendations.

Here is my settings.json:
`{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Car",
  "Vehicles": {
		"Car1": {
			"VehicleType": "PhysXCar",
			"AutoCreate": true,
			"Sensors": {
			    "LidarSensor1": { 
					"SensorType": 6,
					"Enabled" : true,
					"NumberOfChannels": 32,
					"RotationsPerSecond": 10,
					"PointsPerSecond": 100000,
					"X": 0, "Y": 0, "Z": -1,
					"Roll": 0, "Pitch": 0, "Yaw" : 0,
					"VerticalFOVUpper": 65,
					"VerticalFOVLower": -25,
					"HorizontalFOVStart": -80,
					"HorizontalFOVEnd": 80,
					"DrawDebugPoints": true,
					"DataFrame": "SensorLocalFrame"
				}
			}
		}
    }
}
`

OS: Windows 10
Airsim Version: release v1.3.1
Unreal Engine Version: 4.24.3

rajat2004, 2020-08-19:
Could this be related somehow to #2948?

jonyMarino, 2021-07-06:
Hi!
You need to change some things in the code and environment to be able to get that.
Look at this https://github.com/microsoft/AirSim/issues/2948#issuecomment-675355578
and this https://github.com/microsoft/AirSim/issues/3855#issuecomment-874984693

closed 2021-07-06 by jonyMarino