microsoft/AirSim #2726: CS0103: The name `BuildSettingsVersion' does not exist in the current contex

greymaner, 2020-05-26:
Fixing inconsistent case in filenames.
Setting up Mono
~/UnrealEngine/Engine ~/UnrealEngine/Engine/Binaries/Linux
While compiling /home/zhou/AirSim/Unreal/Environments/Blocks/Intermediate/Build/BuildRules/BlocksModuleRules.dll:
/home/zhou/AirSim/Unreal/Environments/Blocks/Source/BlocksEditor.Target.cs(12,32) : error CS0103: The name `BuildSettingsVersion' does not exist in the current context
/home/zhou/AirSim/Unreal/Environments/Blocks/Source/BlocksEditor.Target.cs(12,9) : error CS0103: The name `DefaultBuildSettings' does not exist in the current context
ERROR: Unable to compile source files.
LogInit: Warning: Still incompatible or missing module: Blocks
LogInit: Warning: Still incompatible or missing module: AirSim
LogExit: Preparing to exit.

It come out When I open the blocks with ./UE4Editor /home/xxx/AirSim/Unreal/Environments/Blocks/Blocks.uproject 

How to solve this problem?

rajat2004, 2020-05-26:
UE version isn't mentioned here, which is suspect is the cause of the problem. AirSim supports 4.25, which requires that field, you could probably comment that line out, this is where it was added - https://github.com/microsoft/AirSim/commit/da7dd53ccaf9bbcd2b818dd48ba753196363be6b#diff-ef507fcd2c19937f732794b4c262300aR12

greymaner, 2020-05-27:
@rajat2004 Thanks a lot , my UE version is 4.22, and it is ok after commenting the line out

closed 2020-05-27 by greymaner