microsoft/AirSim #2727: Error at Launch: Failed to load asset class when running my own unreal project

lj8385174, 2020-05-26:
## System Environment
Operating System: Windows 10 64bit 1909
Visual Studio version: 2019, 16.5.4
Unreal Engine version: 4.24.3
Airsim version:  release 1.3.1
CMake version:  3.17.0

## What happened:
When Launching my own unreal project, it prompts that "Failed to load asset class - Class'/AirSim/VehicleAdv/SUV/SuvCarPawn.SuvCarPawn_C'" :
[image]

## Detailed Description 
I am following the [official tutorial](https://github.com/Microsoft/AirSim/blob/master/docs/unreal_custenv.md)  to build my own unreal environment. I can't  find `Landscape Mountains`  in `Learn`  tab of `Epic Game Launcher` so I create a new unreal project in unreal engine with the template `Archvis`:
[image]
I delete all unrelated components, then the world outliner looks like this:
[image]
I carefully follow the instruction in the tutorial, the codes in Visual Studio are compiled correctly, except for some warnings:
```
1>UnrealBuildTool : warning : Library 'Shell32.lib' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
1>UnrealBuildTool : warning : Library 'dinput8.lib' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
1>UnrealBuildTool : warning : Library 'dxguid.lib' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
```
which are also shown in the Airsim's default unreal project `Block`, so I ignore these warnings.

Then I start the game by pressing `Launch` button in Unreal Editor. At the startup menu, it pops up the above message, i.e., "Failed to load asset class - Class'/AirSim/VehicleAdv/SUV/SuvCarPawn.SuvCarPawn_C'". 
This following picture is the logging output screen:
[image]

## Other Attempts
1. The default Unreal project `Block` runs successfully following the same tutorial.
2. I also tried using a quadrotor instead of a vehicle, but it pops up a similar error as follows:
[image]
3. I tried to check the option `Enable on the fly` in `Launch` panel, surprisingly it runs correctly.
4. I press the `Play` button, it also runs correctly.

lj8385174, 2020-05-26:
Finally, I addressed this issue by following the solution in #1310. However, I should also manually copy Airsim .umap file to the content folder.

closed 2020-05-26 by lj8385174

Ckq-Sugar, 2020-06-01:
Hello, I also stuck at starting playing. The issue in #1310 is about packaging. Could you tell me some details about solving the problem? Thanks.

lj8385174, 2020-06-02:
> @lj8385174 Hello, I also stuck at starting playing. The issue in #1310 is about packaging. Could you tell me some details about solving the problem? Thanks.

My problem is stuck in packaging, not playing 😀. So I may not have the experience to help you.

lj8385174, 2020-06-04:
@Ckq-Sugar I recently experienced the same error, while I used `build.cmd  --no-full-poly-car` to build Airsim. After removing this argument, it can run correctly.

Coinstine, 2025-04-23:
hi dude, im new to airsim and have the same problem, could u tell me the detail how u solve the problem？