microsoft/AirSim #2732: Build Errors

00george, 2020-05-28:
I am trying to build the latest version of AirSim on windows using visual studio 2019. I have windows SDK version 10.0.19041.0 but when I go to build the plugin I get 73 errors and 5 warnings.  The errors all seem to be LNK2019: unresolved external symbol. 

`Build FAILED.

"D:\AirSimSimulator\AirSim\AirSim.sln" (default target) (1) ->
"D:\AirSimSimulator\AirSim\DroneShell\DroneShell.vcxproj" (default target) (2) ->
(Link target) ->
  D:\AirSimSimulator\AirSim\DroneShell\\..\AirLib\deps\rpclib\lib\x64\Debug\rpc.lib : warning LNK4272: library machine
type 'x86' conflicts with target machine type 'x64' [D:\AirSimSimulator\AirSim\DroneShell\DroneShell.vcxproj]

"D:\AirSimSimulator\AirSim\AirSim.sln" (default target) (1) ->
"D:\AirSimSimulator\AirSim\HelloDrone\HelloDrone.vcxproj" (default target) (5) ->
  D:\AirSimSimulator\AirSim\HelloDrone\\..\AirLib\deps\rpclib\lib\x64\Debug\rpc.lib : warning LNK4272: library machine
type 'x86' conflicts with target machine type 'x64' [D:\AirSimSimulator\AirSim\HelloDrone\HelloDrone.vcxproj]

"D:\AirSimSimulator\AirSim\AirSim.sln" (default target) (1) ->
"D:\AirSimSimulator\AirSim\DroneServer\DroneServer.vcxproj" (default target) (6) ->
  D:\AirSimSimulator\AirSim\DroneServer\\..\AirLib\deps\rpclib\lib\x64\Debug\rpc.lib : warning LNK4272: library machine
 type 'x86' conflicts with target machine type 'x64' [D:\AirSimSimulator\AirSim\DroneServer\DroneServer.vcxproj]

"D:\AirSimSimulator\AirSim\AirSim.sln" (default target) (1) ->
"D:\AirSimSimulator\AirSim\Examples\Examples.vcxproj" (default target) (8) ->
  D:\AirSimSimulator\AirSim\Examples\\..\AirLib\deps\rpclib\lib\x64\Debug\rpc.lib : warning LNK4272: library machine ty
pe 'x86' conflicts with target machine type 'x64' [D:\AirSimSimulator\AirSim\Examples\Examples.vcxproj]

"D:\AirSimSimulator\AirSim\AirSim.sln" (default target) (1) ->
"D:\AirSimSimulator\AirSim\HelloCar\HelloCar.vcxproj" (default target) (12) ->
  D:\AirSimSimulator\AirSim\HelloCar\\..\AirLib\deps\rpclib\lib\x64\Debug\rpc.lib : warning LNK4272: library machine ty
pe 'x86' conflicts with target machine type 'x64' [D:\AirSimSimulator\AirSim\HelloCar\HelloCar.vcxproj]

"D:\AirSimSimulator\AirSim\AirSim.sln" (default target) (1) ->
"D:\AirSimSimulator\AirSim\DroneShell\DroneShell.vcxproj" (default target) (2) ->
(Link target) ->
  AirLib.lib(RpcLibClientBase.obj) : error LNK2019: unresolved external symbol "public: __cdecl rpc::client::client(cla
ss std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,unsigned short)" (??0clien
t@rpc@@QEAA@AEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@G@Z) referenced in function "public: __cde
cl msr::airlib::RpcLibClientBase::impl::impl(class std::basic_string<char,struct std::char_traits<char>,class std::allo
cator<char> > const &,unsigned short,float)" (??0impl@RpcLibClientBase@airlib@msr@@QEAA@AEBV?$basic_string@DU?$char_tra
its@D@std@@V?$allocator@D@2@@std@@GM@Z) [D:\AirSimSimulator\AirSim\DroneShell\DroneShell.vcxproj]
  AirLib.lib(RpcLibClientBase.obj) : error LNK2019: unresolved external symbol "public: __cdecl rpc::client::~client(vo
id)" (??1client@rpc@@QEAA@XZ) referenced in function "int `public: __cdecl msr::airlib::RpcLibClientBase::impl::impl(cl
ass std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,unsigned short,float)'::`
1'::dtor$0" (?dtor$0@?0???0impl@RpcLibClientBase@airlib@msr@@QEAA@AEBV?$basic_string@DU?$char_traits@D@std@@V?$allocato
r@D@2@@std@@GM@Z@4HA) [D:\AirSimSimulator\AirSim\DroneShell\DroneShell.vcxproj]
  AirLib.lib(RpcLibClientBase.obj) : error LNK2019: unresolved external symbol "public: class nonstd::optional_lite::op
tional<__int64> __cdecl rpc::client::get_timeout(void)const " (?get_timeout@client@rpc@@QEBA?AV?$optional@_J@optional_l
ite@nonstd@@XZ) referenced in function "public: class clmdep_msgpack::v1::object_handle __cdecl rpc::client::call<>(cla
ss std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &)" (??$call@$$V@client@rpc@
@QEAA?AVobject_handle@v1@clmdep_msgpack@@AEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@Z) [D:\AirSi
mSimulator\AirSim\DroneShell\DroneShell.vcxproj]
  AirLib.lib(MultirotorRpcLibClient.obj) : error LNK2001: unresolved external symbol "public: class nonstd::optional_li
te::optional<__int64> __cdecl rpc::client::get_timeout(void)const " (?get_timeout@client@rpc@@QEBA?AV?$optional@_J@opti
onal_lite@nonstd@@XZ) [D:\AirSimSimulator\AirSim\DroneShell\DroneShell.vcxproj]
  AirLib.lib(RpcLibClientBase.obj) : error LNK2019: unresolved external symbol "public: void __cdecl rpc::client::set_t
imeout(__int64)" (?set_timeout@client@rpc@@QEAAX_J@Z) referenced in function "public: __cdecl msr::airlib::RpcLibClient
Base::impl::impl(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,unsign
ed short,float)" (??0impl@RpcLibClientBase@airlib@msr@@QEAA@AEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@
@std@@GM@Z) [D:\AirSimSimulator\AirSim\DroneShell\DroneShell.vcxproj]
  AirLib.lib(RpcLibClientBase.obj) : error LNK2019: unresolved external symbol "public: enum rpc::client::connection_st
ate __cdecl rpc::client::get_connection_state(void)const " (?get_connection_state@client@rpc@@QEBA?AW4connection_state@
12@XZ) referenced in function "public: enum msr::airlib::RpcLibClientBase::ConnectionState __cdecl msr::airlib::RpcLibC
lientBase::getConnectionState(void)" (?getConnectionState@RpcLibClientBase@airlib@msr@@QEAA?AW4ConnectionState@123@XZ)
[D:\AirSimSimulator\AirSim\DroneShell\DroneShell.vcxproj]
  AirLib.lib(RpcLibClientBase.obj) : error LNK2019: unresolved external symbol "private: void __cdecl rpc::client::wait
_conn(void)" (?wait_conn@client@rpc@@AEAAXXZ) referenced in function "public: class std::future<class clmdep_msgpack::v
1::object_handle> __cdecl rpc::client::async_call<>(class std::basic_string<char,struct std::char_traits<char>,class st
d::allocator<char> > const &)" (??$async_call@$$V@client@rpc@@QEAA?AV?$future@Vobject_handle@v1@clmdep_msgpack@@@std@@A
EBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@3@@Z) [D:\AirSimSimulator\AirSim\DroneShell\DroneShell.vcxpr
oj]
  AirLib.lib(MultirotorRpcLibClient.obj) : error LNK2001: unresolved external symbol "private: void __cdecl rpc::client
[... 308 lines cut ...]
@V?$allocator@D@2@@4@V?$shared_ptr@V?$promise@Vobject_handle@v1@clmdep_msgpack@@@std@@@4@@Z) [D:\AirSimSimulator\AirSim
\HelloCar\HelloCar.vcxproj]
  AirLib.lib(RpcLibClientBase.obj) : error LNK2019: unresolved external symbol "private: int __cdecl rpc::client::get_n
ext_call_idx(void)" (?get_next_call_idx@client@rpc@@AEAAHXZ) referenced in function "public: class std::future<class cl
mdep_msgpack::v1::object_handle> __cdecl rpc::client::async_call<>(class std::basic_string<char,struct std::char_traits
<char>,class std::allocator<char> > const &)" (??$async_call@$$V@client@rpc@@QEAA?AV?$future@Vobject_handle@v1@clmdep_m
sgpack@@@std@@AEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@3@@Z) [D:\AirSimSimulator\AirSim\HelloCar\Hel
loCar.vcxproj]
  AirLib.lib(CarRpcLibClient.obj) : error LNK2001: unresolved external symbol "private: int __cdecl rpc::client::get_ne
xt_call_idx(void)" (?get_next_call_idx@client@rpc@@AEAAHXZ) [D:\AirSimSimulator\AirSim\HelloCar\HelloCar.vcxproj]
  AirLib.lib(RpcLibClientBase.obj) : error LNK2019: unresolved external symbol "private: void __cdecl rpc::client::thro
w_timeout(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &)" (?throw_tim
eout@client@rpc@@AEAAXAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@Z) referenced in function "publ
ic: class clmdep_msgpack::v1::object_handle __cdecl rpc::client::call<>(class std::basic_string<char,struct std::char_t
raits<char>,class std::allocator<char> > const &)" (??$call@$$V@client@rpc@@QEAA?AVobject_handle@v1@clmdep_msgpack@@AEB
V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@Z) [D:\AirSimSimulator\AirSim\HelloCar\HelloCar.vcxproj]
  AirLib.lib(CarRpcLibClient.obj) : error LNK2001: unresolved external symbol "private: void __cdecl rpc::client::throw
_timeout(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &)" (?throw_time
out@client@rpc@@AEAAXAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@Z) [D:\AirSimSimulator\AirSim\He
lloCar\HelloCar.vcxproj]
  D:\AirSimSimulator\AirSim\HelloCar\build\x64\Debug\HelloCar.exe : fatal error LNK1120: 9 unresolved externals [D:\Air
SimSimulator\AirSim\HelloCar\HelloCar.vcxproj]

    5 Warning(s)
    73 Error(s)`

rajat2004, 2020-05-28:
Were all the steps in https://github.com/microsoft/AirSim/blob/master/docs/build_windows.md#how-to-use-airsim done? Specifically, Develop Editor & x64?

00george, 2020-05-28:
This is before using airsim and is when trying to build it. The errors are when running build.cmd in Developer Command Prompt for VS 2019

madratman, 2020-07-29:
You are definitely missing a step in the setup. Please re-open if you're still facing issues with current master.

closed 2020-07-29 by madratman