microsoft/AirSim #2739: " Error at startup: invalid map<K, T> key " While Adding New Vehicle

tlgkyn06, 2020-05-30:
Hello ,
I suppose I created vehicle perfectly (-> Imported Fbx mesh with Physics -> Used Vehicle car Blue print -> Created Animation BP accordingly) 

I follow steps at this link : https://carla.readthedocs.io/en/latest/tuto_A_add_vehicle/

but when I try to play; I face with "Error at startup: invalid map<K, T> key" error message . Where did I do wrong? Anybody to help?
 I tried some solutions at : https://github.com/microsoft/AirSim/issues/1506 but nothing works.

OS: Windows 10
Unreal version : 4.24.3
Airsim Version:1.3.1

BrightSoulXYHY, 2020-05-31:
It seems you access vehicle by wrong name when using API

tlgkyn06, 2020-06-01:
Thanks for reply . I found workaround. I added new vehicle Skeletal mesh to current SUV vehicle blueprint.

closed 2020-06-01 by tlgkyn06