microsoft/AirSim #2744: How to implement Reinforcement learning and train model on airsim?

yash1315, 2020-06-02:
I am trying to implement the demo RL code that has been provided by the airsim under RL with a car title.  What are the initial steps to implement the provided code?

rajat2004, 2020-06-04:
Have you looked at https://microsoft.github.io/AirSim/reinforcement_learning/?

yash1315, 2020-06-04:
Yes, I have looked into it. Can you help me with the two basic questions: 
1) how to run the hello_car.py demo code under pythonClient folder?
2) how to implement RL code with Pytorch to train the model? 
Thanks

rajat2004, 2020-06-04:
Please read the API documentation - https://microsoft.github.io/AirSim/apis/
As to implementing in PyTorch, the code and steps and explained in the doc, should be possible to implement something similar in PyTorch

closed 2020-06-06 by yash1315