microsoft/AirSim #2745: Unable to view the Reinforcement Learning tutorial by Ashish Kapoor

amaalo2, 2020-06-03:
Hello, 

I was trying to view the RL tutorial by Ashish Kapoor using the following link:
https://microsoft.github.io/AirSim/docs/reinforcement_learning

However, the page seems empty. Is this the intended behavior? 

[image]

rajat2004, 2020-06-04:
`docs` shouldn't be there in the link - https://microsoft.github.io/AirSim/reinforcement_learning/
[image]

amaalo2, 2020-06-04:
Ok, so the link on the main website points to the wrong place. 
[image]

PR #2658 (2020-09-03): Update docs
1. Add Python API docs
2. Add links for ArduPilot usage (Closes #2686)
3. Update CHANGELOG.md
4. Add more info about Logs in FAQs 
5. Fix indentation
6. Note about installation in C drive (#2697)
7. Fix RL page link (closes #2745)
8. Add  [PyMarkdown extras extension](https://facelessuser.github.io/pymdown-extensions/extensions/extra/) - requires `pip install pymdown-extensions`

I'm not quite sure as to how to link to the html page from the markdown files other than just having it in the nav bar and mentioning in the markdown to go to them through the github.io page nav bar. Bette [... cut]
closed 2020-09-03 by madratman