microsoft/AirSim #2746: Having the drone face the flight direction using moveToPositionAsync

vguillet, 2020-06-04:
Good afternoon,

I have been wondering how I could make sure that the drone would face the flight direction when using the moveToPositionAsync command, I managed to achieve that using moveOnPathAsync by setting airsim.DrivetrainType.ForwardOnly and airsim.YawMode(False, 0), 20, 1), but that doesnt seem to work with moveToPositionAsync, any suggestions?

Cheers!

sy300302, 2020-06-05:
I got the same problem too! The ForwardOnly mode seems not working, the api returns quickly but the drone not moving.

vguillet, 2020-06-11:
Any solutions?

DanielGilo, 2020-07-09:
Having the same issue using c++ API.

Rick-Weng, 2021-09-27:
This is my code,it work well.
client.moveOnPathAsync(v_path_list, 20, 20000,airsim.DrivetrainType.ForwardOnly, airsim.YawMode( False,0))
ps:when u set" forwardonly",YawMode must be False.