microsoft/AirSim #2747: UnrealEngine install error on Mac

hectorperez, 2020-06-04:
Hi!

I'm trying to install UnrealEngine & AirSim on my Mac [1] following [2] without Docker.

I've done:
```
git clone -b 4.24 git@github.com:EpicGames/UnrealEngine.git
cd UnrealEngine
./Setup.sh
./GenerateProjectFiles.sh
```

The command `./GenerateProjectFiles.sh` returns `The domain/default pair of (com.epicgames.ue4, MonoAOT) does not exist`:
```
Setting up Unreal Engine 4 project files...
Running bundled mono, version: Mono JIT compiler version 5.16.0.220 (2018-06/bb3ae37d71a Fri Nov 16 17:12:11 EST 2018)
2020-06-04 18:35:04.063 defaults[59762:702064] 
The domain/default pair of (com.epicgames.ue4, MonoAOT) does not exist
Generating data for project indexing... 100%
Writing project files... 100%
```

Then, when I run `make` I get the following error - I suppose due to the failing `./GenerateProjectFiles.sh`: 
```
make: *** No targets specified and no makefile found.  Stop.
```

Any idea what I could do? Thank you!

Cheers,
Hector
[1]
MacOS Catalina
Version 10.15.5
MacBook Pro (Retina, 13-inch, Early 2015)
Processor 3.1 GHz Dual-Core Intel Core i7
Memory 16 GB 1867 MHz DDR3
Startup Disk Macintosh SSD
Graphics Intel Iris Graphics 6100 1536 MB

[2] https://microsoft.github.io/AirSim/build_linux

rajat2004, 2020-06-05:
It's not required to build UE4 from scratch on Mac, see https://github.com/microsoft/airsim/blob/master/docs/build_linux.md#macos---download-unreal-engine

Sorry for this, changes to get the github pages site auto-build from master branch are on the way, until then I would suggest for up-to-date info, check the .md files rather than the github.io pages. It will be the same for most but can vary in places like above

hectorperez, 2020-06-05:
Aha, thank you, @rajat2004

closed 2020-06-05 by hectorperez

kyoa, 2021-03-04:
> It's not required to build UE4 from scratch on Mac, see https://github.com/microsoft/airsim/blob/master/docs/build_linux.md#macos---download-unreal-engine
>
> Sorry for this, changes to get the github [...]

Finally, I'm trying to compile for Linux on Mac, met the same problem. What should I do?