microsoft/AirSim #2748: Computer Vision mode not working for Africa Linux binary

vegnesh, 2020-06-04:
# Read This First

## If you are reporting a bug
* Make sure to write **all reproduction steps**
* Include full error message in text form
* Search issues for error message before filing issue
* Attach screenshot if applicable
* Include code to run if applicable

## If you have question
* Add clear and concise title
* Add OS, AirSim version, Python version, Unreal version if applicable
* Include context on what you are trying to achieve
* Include details of what you already did to find answers

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------
My settings.json is 
`{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "ComputerVision"
  
}`

When I launch the Africa airsim binary with the above settings. the following error occurs:
```
[2020.06.04-17.40.46:627][  0]LogStreaming: Error: Couldn't find file for package /AirSim/Blueprints/BP_ComputerVisionPawn requested by async loading code. NameToLoad: /AirSim/Blueprints/BP_ComputerVisionPawn
[2020.06.04-17.40.46:627][  0]LogStreaming: Error: Found 0 dependent packages...
terminating with uncaught exception of type std::invalid_argument: Failed to load asset class - Class'/AirSim/Blueprints/BP_ComputerVisionPawn.BP_ComputerVisionPawn_C'
Signal 6 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554
```

I am running Ubuntu 18.04. Any pointers on how to fix this?

rajat2004, 2020-06-05:
Hmm, strange, Car & Multirotor work in Africa, & CV mode works in Blocks binary. Might be a problem in packaging Africa
FYI @madratman @saihv

jonyMarino, 2021-05-25:
Hi and welcome @vegnesh! We have testes this in the last version https://github.com/microsoft/AirSim/releases
Good luck!

closed 2021-05-25 by jonyMarino