microsoft/AirSim #2750: Open Blocks.uproject error

blyspyder, 2020-06-05:
os : ubuntu18.04
python version : 3.7
airsim version : v1.2.8
Unreal version: 4.18.3
When I open the "Blocks.uproject", I got the error message:
[image]
What can I do for this.

rajat2004, 2020-06-05:
Press Yes to build, it's a common one due tk UE version difference
Note that UE 4.18 mostly won't work with the latest AirSim master, minimum is 4.22 and recommend is 4.24

blyspyder, 2020-06-05:
> Press Yes to build, it's a common one due tk UE version difference
> Note that UE 4.18 mostly won't work with the latest AirSim master, minimum is 4.22 and recommend is 4.24
I press yes and get the error

blyspyder, 2020-06-05:
> Press Yes to build, it's a common one due tk UE version difference
> Note that UE 4.18 mostly won't work with the latest AirSim master, minimum is 4.22 and recommend is 4.24
When I press yes I got this error
[image]

rajat2004, 2020-06-05:
As mentioned, 4.18 is no longer supported, you'll need to upgrade to 4.22 or 4.24
If you want to use 4.18, then checkout an older commit in master

blyspyder, 2020-06-05:
> Press Yes to build, it's a common one due tk UE version difference
> Note that UE 4.18 mostly won't work with the latest AirSim master, minimum is 4.22 and recommend is 4.24

How can I update my UE version in ubuntu18.04. I can't find any information about it.

rajat2004, 2020-06-05:
Follow the steps in https://github.com/microsoft/AirSim/blob/master/docs/build_linux.md to install 4.24

blyspyder, 2020-06-05:
> Follow the steps in https://github.com/microsoft/AirSim/blob/master/docs/build_linux.md to install 4.24

Thank you very much, I will try it now.

closed 2020-07-29 by madratman