microsoft/AirSim #2751: Setting changing trace color using python

vguillet, 2020-06-05:
Good afternoon, I was wondering if there was a way to adjust the trace color mid run using python

Cheers! Loving airsim so far!

rajat2004, 2020-06-05:
Try using [`simSetTraceLine`](https://github.com/microsoft/AirSim/blob/master/PythonClient/airsim/client.py#L305)

vguillet, 2020-06-05:
Thanks for the swift reply! As a follow up, do I call it on client? It doesnt seem to appear in the class definition? (Also, what parameter is used to adjust the color?)

rajat2004, 2020-06-05:
All methods are to be called using a client if that's what you're asking - `client.simSetTraceLine(...)`
Seems like this was added after the current latest `airsim` package was released (`1.2.8`)
You can put your code in one of the folders like `Pythonclient/multirotor` and add `import setup_path` before doing `import airsim` to use the master branch APIs

Also see [set_trace_line.py](https://github.com/microsoft/AirSim/blob/master/PythonClient/multirotor/set_trace_line.py) for usage

vguillet, 2020-06-05:
Great, thank you!

closed 2020-06-05 by vguillet