microsoft/AirSim #2753: How to Create and setup Custom Environment in Linux

thejeshk, 2020-06-06:
Hello Everyone, 

Does anyone know how to setup custom environments in Linux ? I have followed the steps -> https://microsoft.github.io/AirSim/build_linux/  

After this step I am supposed to set up Unreal Environment there are two options 
Option 1: built-in Blocks environment 
Option 2: Create your own Unreal Environment 

I need Option 2. I already have a project created using Microsoft Windows 10 with UE 4.22. 

I am not sure what exactly this means: 
Note for Linux users: There is no Eoic games launcher for Linux which means that if you need to create custom environment, you will need Windows machine to do that. Once you have unreal project folder, just copy it over to your Linux machine. 

I just copied my Project folder 📁 but there is no .uproject file in it. 

Can anyone help me with this who has successfully built custom environment in Linux? 

Thank you 🙂

madratman, 2020-07-29:
" Note for Linux users: There is no Epic games launcher for Linux which means that if you need to create custom environment, you will need Windows machine to do that. Once you have unreal project folder, just copy it over to your Linux machine."

By this, we are referring to the scenario when someone wants to use environments which can be downloaded from the UE marketplace. 
If so, they have to download the environment from UE Marketplace in windows, and then copy the environment to their linux machine. 

For ensuring that a custom environment can compile, you should ensure that you're matching the structure of Blocks project exactly. 

This entails:
- ensuring that Plugins/AirSim is copied over to the custom env folder (your env should match the directory layout of Blocks post `./update_from_git.sh`
- If your env doesn't have a uproject file, make one by referring to Blocks.uproject
- Also ensure, you match the contents of Blocks\Source folder (https://github.com/microsoft/AirSim/tree/master/Unreal/Environments/Blocks/Source), and you can find-and-replace `Blocks` in the contents and names of these files with the name of your UE project.

closed 2020-07-29 by madratman

andrewjong, 2021-08-20:
@madratman thank you fellow AirLabber!

Flautdn, 2022-03-24:
Hi. Thanks will reply if in case I see any issues in the above integration

Flautdn, 2022-03-28:
ERROR: Could not find definition for module 'test2' (referenced via test2Editor.Target.cs)

Flautdn, 2022-03-28:
When compiling with a custom made environment. After adding source folder as per blocks and making changes in .project, I get the above error