microsoft/AirSim #2755: Error of build.sh on mac

dingkailv, 2020-06-06:
[ 95%] Built target HelloCar
[ 98%] Built target DroneShell
Undefined symbols for architecture x86_64:
  "___darwin_check_fd_set_overflow", referenced from:
      mavlinkcom_impl::UdpSocketImpl::pollin(unsigned int) in libMavLinkCom.a(UdpSocketImpl.cpp.o)
      mavlinkcom_impl::UdpSocketImpl::pollout(unsigned int) in libMavLinkCom.a(UdpSocketImpl.cpp.o)
ld: symbol(s) not found for architecture x86_64
clang-8: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [output/bin/AirLibUnitTests] Error 1
make[1]: *** [AirLibUnitTests/CMakeFiles/AirLibUnitTests.dir/all] Error 2
make: *** [all] Error 2

rajat2004, 2020-06-06:
This is a known bug in XCode and is being discussed in #2559, it'll be better to keep it on the same topic

dingkailv, 2020-06-07:
> 这是XCode中的一个已知错误，将在[＃2559中](https://github.com/microsoft/AirSim/issues/2559)进行讨论，最好将其保留在同一主题上

Thank you.

closed 2020-06-07 by dingkailv