microsoft/AirSim #2756: New Landscape map

KabirovIrik, 2020-06-07:
Hello.
I have last version of airsim. And I want to use old version of Landscape Mountains, because it have powerlines. Airsim api don't work in old landscape map.
I wanted to use map from this instruction https://github.com/microsoft/AirSim/blob/master/docs/unreal_custenv.md, but now landscape mountains missed from Learning tab.
What can I do to use old version landscape map with new version Airsim?

rajat2004, 2020-06-07:
Latest release can be downloaded from https://github.com/microsoft/AirSim/releases, in each release there are several different env binaries including LandscapeMountains

KabirovIrik, 2020-06-07:
> Latest release can be downloaded from https://github.com/microsoft/AirSim/releases, in each release there are several different env binaries including LandscapeMountains

I know, but New version map don't has powerlines

rajat2004, 2020-06-07:
Can't do anything about LandscapeMountains not being available in the MArketplace, but if you do get an older version of the project from somewhere, then you can follow the steps in the doc to add the latest AirSim plugin to it. Note that older project might not work with newer versions of Airsim, which requires atleast 4.22 & recommended is 4.24.
IIRC there were powerlines in Neighbourhood env, not sure though

closed 2020-07-29 by madratman