microsoft/AirSim #2757: Using only the API of AirSim without UE4 (graphical side)

WilliamPsc, 2020-06-07:
Hi everyone,

I'm a student at the university and for a project in computer science I need to simulate an accelerometer & gyroscope (IMU) for a simulated drone in C++. I found AirSim for that and I think it could be good to use this library. My problem is that I didn't find any solution to use only the physical engine of AirSim and not everything; I don't need the server side, the graphical side etc just the physical to simulate the sensors and the course of the drone.

I'm working on Windows OS and the last version of AirSim.

Thanks in advance for your answer.

rajat2004, 2020-06-08:
You could try using [`NoDisplay`](https://github.com/microsoft/AirSim/blob/master/docs/settings.md#viewmode) ViewMode to disable the main rendering

madratman, 2020-07-29:
You can compile AirLib as a standalone project, see https://github.com/microsoft/AirSim/tree/master/AirLibUnitTests and  https://github.com/madratman/airsim_standalone/ for reference

closed 2020-07-29 by madratman