microsoft/AirSim #2762: Could not find function 'getImuData' with argument count 2 while running older LandscapeMountains environment with latest Airsim version

pmff96, 2020-06-09:
I'm trying to visualize LiDAR data in LandscapeMountains environment, Linux v.1.2.0 (I realize there is a newer version, but it doesn't have the powerlines or the transmission towers, which are the main focus of my project) in Rviz, following the ROS tutorial. My OS is Ubuntu 18.04 LTS, with the latest AirSim version and Unreal Engine 4.24, Python version is 3.5, although it doesn't seem to be a Python specific issue. The Settings.json file is exactly the same as the front_stereo_and_center_mono.json present in the ROS tutorial. 

The steps to reproduce the issue are stated below:

1. Start LandscapeMountains environment binary
2. roslaunch airsim_ros_pkgs airsim_node.launch, which results in the following error [image]
3. roslaunch airsim_tutorial_pkgs front_stereo_and_center_mono.launch, which results in the following errors 
[image]

When I follow the same steps for the Blocks environment it works fine, but it fails for the LandscapeMountains (with powerlines) environment. I'd just like to know if I'm missing something and/or there's a solution for this issue. Thanks in advance!

rajat2004, 2020-06-09:
`v1.2.0` is a very old release, and the ROS wrapper and APIs used in that were probably added afterwards. You *could* checkout the tag for the release and try to use the APIs present at that point in the repo, but there's no guarantee if that would work

When you say that it works for Blocks env, you mean the latest ROS wrapper works with Blocks `v1.2.0`, but doesn't for LM?? Or just that it works with the latest Blocks binary?

pmff96, 2020-06-09:
I meant the latest Blocks binary, didn't actually try the v.1.2.0 Blocks. I will try to use the APIs present at that time as suggested to see if it works, because unfortunately the latest LandscapeMountains environment, which would probably work without issues, does not have the power lines and transmission towers.

rajat2004, 2020-06-09:
Now that I went back and checked the commit history of the ROS wrapper, it was added after `v1.2.0` release, so that'll be a problem. Rewriting the wrapper to use the old APIs might not work or could reduce the utility of actually using that
An alternative way would be to create or search for assets related to powerlines, etc which are required and add that to your custom env. This will also give a large amount of flexibility in capturing the type of data collected

closed 2020-06-09 by pmff96