microsoft/AirSim #2764: Issue with running airsim in the unreal editor using a python script

nitinjain442, 2020-06-09:
When running the airsim executable file on its own without using the unreal editor and then running the python script using the cmd prompt, the car starts moving in the blocks files immediately and doesn't lag at all.

However, when running the airsim executable file through the unreal editor and when i run the python script from the cmd prompt, it connect and there is connection, however, the car doesn't move at all with the instructions in the python script and lags a bunch and waits until the script has been closed to do something. 

What could be the problem? is there a terminal in unreal that you use that makes it faster?

rajat2004, 2020-06-10:
Is the option `Use Less CPU when in background` unchecked? It's present in Edit ->Editor Preferences and search CPU.
Which script are you using? And what are the settings?

nitinjain442, 2020-06-10:
No, i have made sure it is checked. I am using the hello_car.py python script that comes with the Blocks file from airsim in the car folder which is in the pythonclient folder. 

Settings of what? I am just running the script from the terminal and then checking the output in the unreal editor right now.

rajat2004, 2020-06-10:
I mean the option should be *unchecked*. By default it's checked, which we don't want unless really required
By settings, I mean the settings specfied in `settings.json` in `Documents/AirSim`

nitinjain442, 2020-06-10:
It is working now that I have unchecked the use less cpu option :) How does this make it faster though?

rajat2004, 2020-06-10:
When running scripts, you're changing windows to go to the terminal, etc. and they remain in the foreground, say either to be interactive, or just to observe the output, etc. UE is meant for things like games, where it's supposed to be the main thing, and other programs run in the background.
Using less CPU in background will delay messages, and overall running of the Physics will also slow down to a bare minimum along with the graphics. So you don't want to have it enabled unless required

If the problem has been fixed, issue can be closed

nitinjain442, 2020-06-10:
ohh okay. Thanks!!

closed 2020-06-10 by nitinjain442