microsoft/AirSim #2767: how can i get Environments source code?

kimwin2, 2020-06-11:
Hi.
I'm trying to use Airsim to apply autonomous car environment. 
i just downloaded release environment such as blocks, cities which is binarized version. 
I have a question that how can i get a source code version? Is it possible to get it?

thanks.

rajat2004, 2020-06-11:
Blocks source code included in the repo, and is the default env for testing
Envs can be downloaded from Marketplace - [Neighbourhood](https://www.unrealengine.com/marketplace/en-US/product/modular-neighborhood-pack), [City](https://www.unrealengine.com/marketplace/en-US/product/downtown)  and lot's more

madratman, 2020-07-29:
Apart from the above, the envs are closed source.

closed 2020-07-29 by madratman