microsoft/AirSim #2768: Xbox One controller drone always going up

agentfeux, 2020-06-11:
Hi,
I have a issue when i only use Airsim with unreal, my drone is always trying to go up. It's not doing it when i use it with Ardupilot and Qgroundcontrol.

Thanks for helping me with this problem :)

[The controller](https://www.amazon.ca/PDP-Wired-Controller-Xbox-One/dp/B073X47SG4/ref=sr_1_8?crid=2TCRVSSI573T2&dchild=1&keywords=pdp+xbox+one+controller&qid=1591900390&sprefix=pdp+x%2Caps%2C167&sr=8-8)

rajat2004, 2020-06-12:
When using AirSim, are you using SimpleFlight or ArduPilot? When working on the AP-Airsim integration, I had added the fields for using RC, where the RC inputs come from Airsim. My only testing was some dummy hardcoded values in the messages, since I didn't have any RC with me at the time. So I pretty much have no idea if it works properly or not when using with different RCs.

Have you calibrated the controller with QGC? Checking if values being sent to AP make sense would be required, for that you could either use something like Wireshark to capture the UDP packets being sent, and the values should be easily visible.
Else could add a print statement here - https://github.com/ArduPilot/ardupilot/blob/master/libraries/SITL/SIM_AirSim.cpp#L308

agentfeux, 2020-06-12:
Only when i am using SimpleFlight, the AP-Airsim integration is working fine and i can calibrate my controller with QGC so no problem with that. I am testing scripts and the quickest way to do it is with SimpleFlight but it's hard to control when somehow the drone move up on it own.
For info this is my setting.json:
`{
    "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings_json.md",
    "SettingsVersion": 1.2,

    "SimMode": "Multirotor",

     "Vehicles": {
		"Drone1": {
			"VehicleType": "simpleflight",
			"AutoCreate": true,
			"RC": {
				"RemoteControlID": 0,
				"AllowAPIWhenDisconnected": false
			  },
			"Sensors": {
			    "LidarSensor1": { 
					"SensorType": 6,
					"Enabled" : true,
					"NumberOfChannels": 2,
					"RotationsPerSecond": 10,
					"PointsPerSecond": 10000,
					"X": 0, "Y": 0, "Z": 0,
					"Roll": 0, "Pitch": 0, "Yaw" : 0,
					"VerticalFOVUpper": 0,
					"VerticalFOVLower": 0,
					"HorizontalFOVStart": 0,
					"HorizontalFOVEnd": 0,
					"DrawDebugPoints": true,
					"DataFrame": "SensorLocalFrame"
				}
			}
		}
    }
}
`

rajat2004, 2020-06-12:
Hmm, that's pretty strange. Glad to know it's working correctly with AP.
Not sure as to why this would happen with SimpleFlight, are the RC value being displayed in the viewport correct? Otherwise you could also write a small script to get the RC values and print it on the terminal, something like -
```
while(True):
    print(client.getMultirotorState().rc_data)
```

agentfeux, 2020-06-12:
Yeah they are dispaying on the screen
[image]

if you want more details:
```
<instance> {   u'is_initialized': True,
    u'is_valid': True,
    u'left_z': -0.999969482421875,
    u'pitch': -0.028519153594970703,
    u'right_z': -0.999969482421875,
    u'roll': -0.012436091899871826,
    u'switches': 0,
    u'throttle': 0.7625049352645874,
    u'timestamp': 0,
    u'vendor_id': u'',
    u'yaw': 0.012283563613891602}
```
For the controller driver
[image]
[image]

agentfeux, 2020-06-12:
Also, my coworker use a PS4 controller and he have the same problem.

Both controller are wired.

rajat2004, 2020-06-13:
@agentfeux Could you try #2770? I'm not sure, just a try from looking at the code
From the calibration win, Axis 2 seems to be strange since it has 0 for everything. Could you also check if the TRPY values being reported are actually in sync with the expected values (both on master and maybe the PR)

agentfeux, 2020-06-14:
Thanks you the fix work for me ! instead of always going up, its slowly going down and i just need to push the joystick down to go up.

> Could you also check if the TRPY values being reported are actually in sync with the expected values (both on master and maybe the PR)

Not sure what you mean about the TRPY master and PR but the new value i got new value.
[image]

rajat2004, 2020-06-15:
I've pushed a commit to reverse the Throttle input, that should probably fix the up-down problem.
By sync, meant that the values are the correct direction, or some scaling issues, etc.
Also, I'm not sure which axis does the Throttle stick correspond to, I'm assuming 1. From the calibration wizard, the max min values seem correct

PR #2770 (2021-03-01): XBox Controller fix
Fixes #2768 
Pending testing from @agentfeux

Zero2One conversion is being done here - https://github.com/microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/PawnSimApi.cpp#L227
Before this PR, it was changing an already 0-1 value to 0-1, so it introduced a positive offset

This change won't affect Win since it's kept behind #ifdefs
closed 2021-03-01 by zimmy87