microsoft/AirSim #2772: How to extract Gimbal Orientation?

Omar-Elmofty, 2020-06-13:
I am using active gimbal stabalization in airsim to stabalize my camera while flying (below are my camera settings that I am using). I'd like to extract the gimbal orientation (relative to the drone's base frame) in realtime while the vehicle is flying, I basically want to know where the camera is relative to the body of the drone at all time. I am wondering how to do that?

 "front_right_custom": {
          "CaptureSettings": [
            {
              "PublishToRos": 1,
              "ImageType": 0,
              "Width": 672,
              "Height": 376,
              "FOV_Degrees": 90,
              "TargetGamma": 1.5
            }
          ],
          "Gimbal": {
            "PublishToRos": 1,
            "Stabilization": 1,
            "Pitch": -30, "Roll": 0
          },
          "X": 0, "Y": 0.06, "Z": 0.2,
          "Pitch": 0, "Roll": 0, "Yaw": 0

rajat2004, 2020-06-14:
Have you tried using [`simGetCameraInfo`](https://github.com/microsoft/AirSim/blob/master/docs/image_apis.md#camera-apis)

closed 2020-06-27 by Omar-Elmofty

ZGX010, 2021-02-02:
you can try this:

change

> "Gimbal": { "PublishToRos": 1, "Stabilization": 1, "Pitch": -30, "Roll": 0 }
to 

> "Gimbal": { "PublishToRos": 1, "Stabilization": 1 }