microsoft/AirSim #2773: How to use SimpleFlight outside of AirSim?

alonfaraj, 2020-06-14:
OS - Win64
AirSim - master

I want to use simple_flight in a project **outside** of airsim and unreal which is possible according to the documentation:
_"This means there is no special build required to compile simple_flight. You just copy its source code to any project you wish and it just works."_

I compiled **AirLib** and run the **SimpleFlightTest** under **AirLibUnitTests** project, but it always failing.
The crash is inside
`msr.airlib.AirSimSimpleFlightCommon.toAxis3r(const Vector3r& vec)`
 which get `vec = NULL` and output an error:
`Exception thrown: read access violation.
**Eigen::DenseCoeffsBase<Eigen::Matrix<float,3,1,0,3,1>,0>::x**(...) returned nullptr.`

and also got multiple console prints:
`Exception occurred while updating world: reset() must be called first before update()`

Does anyone can guide me how to use in a standalone project?
Is it possible?

madratman, 2020-07-29:
https://github.com/madratman/airsim_standalone/ a bit outdated, but does this help?

jdow749, 2020-10-02:
I'm having the exact same issue (also on OS - Win64, AirSim - master). In fact all of the tests in AirLibUnitTests fail.  Has anyone looked into this?  No, the linked repo didn't help, all that is doing is pulling the hello_drone project into a separate repo (missing the point).  The goal it so be able to use AirSim without any 3D environment (i.e. just get access to the drone physics without needing Unreal or Unity).

jdow749, 2020-10-05:
To follow up, I've figured out how to fix the crashes (add the following lines at [line 48 of SimpleFlightTest.hpp] (https://github.com/jdow749/HiveAirSim/blob/edec6278a661d56df3f4677359d7f94f0011e9a8/AirLibUnitTests/SimpleFlightTest.hpp#L48))

    api->setSimulatedGroundTruth(&initial_kinematic_state, environment.get());
    api->reset();
    kinematics->reset();

It appears that the api needs to be initialized and reset and the kinematics needs to be reset.  Unfortunately that doesn't make the drone do anything, it just stays in the same position after the Takeoff command times out (same with the moveto command) and no errors or feedback in the status messages. Any ideas what might be going on?

To try and debug I started rolling back to prior commits that touched the file and found a working version in [commit 170cf69d95808533dc2800decdcc58da996c6342](https://github.com/microsoft/AirSim/commit/170cf69d95808533dc2800decdcc58da996c6342). This is from Feb 13, 2018 so the test hasn't been working in a long time.  Anyway the code is pretty different therefore I'm not sure what is responsible for breaking things.  After a few tweaks to the VS project (these are targeted to VS2015) I was able to run and get the following output good looking :

requestApiControl was successful
Vehicle is already armed
lookahead = 4.000000, adaptive_lookahead = 0.000000
commandVelocity -3.167968, -3.167968, -2.219899, 0.000000
API call timed out, entering hover mode
commandVelocity -3.167960, -3.167960, -2.219922, 0.000000
commandVelocity -3.167811, -3.167809, -2.220351, 0.000000
...
commandVelocityZ -3.264298, -3.319396, -5.000000, 0.000000
commandVelocityZ -3.252239, -3.307847, -5.000000, 0.000000
commandVelocityZ -3.231284, -3.287475, -5.000000, 0.000000
commandVelocityZ -3.195342, -3.252220, -5.000000, 0.000000
commandVelocityZ 0.000000, 0.000000, -5.000000, 0.000000
commandVelocityZ 0.000000, 0.000000, -5.000000, 0.000000
commandVelocityZ 0.000000, 0.000000, -5.000000, 0.000000
commandVelocityZ 0.000000, 0.000000, -5.000000, 0.000000
commandVelocityZ 0.000000, 0.000000, -5.000000, 0.000000
segment 0 done: x=0.000000, y=-1.496336, z=-1.496336
overshoot=0.053079
commandVelocityZ 0.000000, 0.000000, -5.000000, 0.000000
API call timed out, entering hover mode

I'd prefer not to have to fork a 3 year old version of the code if possible,  does anyone have any ideas of what needs to be modified in HEAD to get the test working?

Thanks!

jonyMarino, 2020-10-06:
Hi @jdow749, may be #2558 can help you

jdow749, 2020-10-06:
jonyMarino, thanks for the suggestion! Looks like [commit 9c4e59d1a2b371ebc60cdc18f93b06cbe3e9d305](https://github.com/microsoft/AirSim/pull/2558/commits/9c4e59d1a2b371ebc60cdc18f93b06cbe3e9d305) does show a little bit more about how the params are loaded from the json settings file, but unfortunately it doesn't do anything with the drone once it is created (unless I'm missing something).  

As an update to debugging, I noticed that in the version of the SimpleFlightTest at head modified with the init/reset commands specified earlier in this thread that I'm seeing motion if I look at the kinematic body by adding the following lines:

    StateReporter reporter;
    kinematics->reportState(reporter);

This is opposed to checking the multirotor state which I was doing before which shows no movement:
 
 Vector3r pos = api->getMultirotorState().getPosition();

The positions reported by the kinematics still don't look correct (as the take off moves really high and the moveto command never converges to anything close to the target), so there is still something wrong.  It seems strange that the kinematics would report movement, but nothing would show up in the api which should be using the same kinematics object under the hood via the vehicle.

jrtcppv, 2021-05-25:
Using v1.4.0 I was unable to get standalone physics working either. Here is the simple example I wrote, which compiles but hangs indefinitely once I command the vehicle to do anything:

```cpp
  1 #include <iostream>
  2
  3 #include "common/SteppableClock.hpp"
  4 #include "MavLinkVehicle.hpp"
  5 #include "physics/FastPhysicsEngine.hpp"
  6 #include "vehicles/multirotor/MultiRotorParamsFactory.hpp"
  7 #include "vehicles/multirotor/MultiRotorPhysicsBody.hpp"
  8
  9 int main() {
 10   using namespace msr::airlib;
 11
 12   std::shared_ptr<SteppableClock> clock = std::make_shared<SteppableClock>(1E-3f);
 13   ClockFactory::get(clock);
 14
 15   std::unique_ptr<MultiRotorParams> params = MultiRotorParamsFactory::createConfig(
 16     AirSimSettings::singleton().getVehicleSetting("SimpleFlight"),
 17     std::make_shared<SensorFactory>());
 18   auto api = params->createMultirotorApi();
 19
 20   auto initial_kinematics = Kinematics::State::zero();
 21   initial_kinematics.pose = Pose::zero();
 22   initial_kinematics.pose.position.z() = -1;
 23   Environment::State initial_environment;
 24   initial_environment.position = initial_kinematics.pose.position;
 25   Environment environment(initial_environment);
 26   Kinematics kinematics(initial_kinematics);
 27   api->setSimulatedGroundTruth(&initial_kinematics, &environment);
 28   api->reset();
 29   kinematics.reset();
 30
 31   // Set up multirotor body.
 32   MultiRotorPhysicsBody body(params.get(), api.get(), &kinematics, &environment);
 33   body.reset();
 34
 35   // Set up physics engine.
 36   FastPhysicsEngine physics;
 37   physics.insert(&body);
 38   physics.reset();
 39
 40   api->enableApiControl(true);
 41   api->armDisarm(true);
 42   // The commented lines below cause an indefinite hang.
 43   //api->takeoff(0);
 44   //clock->sleep_for(2.0f);
 45   //api->moveToPosition(-5, -5, -5, 5, 1e3, DrivetrainType::ForwardOnly, YawMode(false, 0), -1, 0);
 46
 47   for (unsigned int i = 0; i < 1000000; i++) {
 48     clock->step();
 49     environment.update();
 50     body.update();
 51     physics.update();
 52     ++i;
 53     auto pose = body.getKinematics().pose;
 54     Quaternionr orientation = pose.orientation;
 55     auto position = pose.position;
 56     std::cout << "POSITION: " << position << std::endl;
 57   }
 58 }
```
Am I doing something wrong in the setup? It would really help if the simple flight example was working.