microsoft/AirSim #2776: Difficulty in running DQNcar.py file

yash1315, 2020-06-16:
Hey guys,
Currently, I am trying to implement the DQNcar.py file on my system and I left my system to run overnight (7-8 hr) but still, I could not see the output. My question is, Does anyone know approximately how long will it take to run the DQNcar.py file with this specification?
intel i5
8 GB RAM
intel 520 HD graphic card

And what can I do to boost this process? Also, Can I use Azure NC/NV to execute this file and produce some results? If not then what is the best possible solution?

jonyMarino, 2020-08-27:
Hi. The DQN script has some issues, wait for an update. It is possible to run on azure. If you need graphical output you would need a NV

zimmy87, 2021-01-07:
possibly related to #1009

zimmy87, 2021-01-15:
Hi @yash1315, thank you for reporting this issue. This script was recently removed in #3215 and replaced with a new script at \AirSim\PythonClient\reinforcement_learning\dqn_car.py that uses the new airgym module. I tested the new script locally and it runs fine for me. Can you retest with the new script and confirm whether you are still seeing this issue or not?

closed 2021-01-15 by zimmy87