microsoft/AirSim #2779: angular and linear velocity confusion

WouterJansen, 2020-06-18:
Why is the groundtruth kinematics of the Unreal plugin using absolute linear velocity and relative angular velocity? Why this difference? Was this done for a particular reason or an oversight?

for the IMU sensor you correct the linear velocity from absolute to relative but when using the API to get the kinematic state (`simGetGroundTruthKinematics()`) you get absolute linear and relative angular velocity respectively which seems odd to me.

rajat2004, 2020-06-20:
The relevant code for IMU is here - https://github.com/microsoft/AirSim/blob/master/AirLib/include/sensors/imu/ImuSimple.hpp#L54-L60

I'm confused about the angular velocity part, according to the code, the values returned by IMU & ground truth kinematics should be same right? Linear acc is being converted to body frame as seen.
And the problem is that GroundTruthKinematics ang_vel is in relative frame rather than world frame

I'm not very good at Physics so sorry for any confusion

WouterJansen, 2020-06-20:
Well the main issue is linear and angular being in different reference frames. They should be in either world or relative, not both. That just leads to confusion in my opinion. 

IMU should always be relative, so the conversion takes care of that so no issue there.

I personally don't know if the best this for GroundTruthKinematics is relative or world. a world angular velocity makes no sense I think, it's always relative. So I think linear should be changed as well to relative.

I'm not sure about acceleration, haven't checked what those are in.