microsoft/AirSim #2782: Tracking user-defined state by modifying existing Python APIs

nicocarb, 2020-06-18:
I am going to throw this here in case anybody has any ideas. I have added a parameter to KinematicsState that tracks a payload_position at any time within  `RpcLibAdapatorsBase.hpp ` and incorporated it in a few other places such as `CommonStructs.hpp` , `AirSimSimpleFlightEstimator.hpp`, and `AirSimSimpleFlightCommon.hpp`. My goal is to be able to access this value when I call `state = client.getMultirotorState()` from my Python script by then calling state.kinematics_estimated.payload_position; however, I can only see the value get updated when I set a debug point within `RpcLibAdapatorsBase.hpp` . Once it binds and returns it in `MultirotorRpcLibServer.cpp`  (line 166), it gives me an error stating AttributeError: 'KinematicsState' object has no attribute 'payload_position' despite my watch variable in Visual Studios showing a payload_position attribute.
 
Although I don't understand communication protocol too well, if I delete the "payload_position" attribute in the KinematicsState Class MSGPACK_DEFINE_MAP command within `RpcLibAdapatorsBase.hpp` , everything compiles and runs fine (but I can't see the payload_position). My description is likely quite confusing (as I too am uncertain as to what is going on), so I can clarify any questions, but I was wondering if anybody had any idea on where to look deeper into to allow me to track my user defined payload_position variable. 

`MSGPACK_DEFINE_MAP(position, orientation, payload_position, linear_velocity, angular_velocity, linear_acceleration, angular_acceleration);
 `

rajat2004, 2020-06-18:
Did you add your `payload_position` in all the functions in [`MultirotorRpcLibAdaptors.hpp`](https://github.com/microsoft/AirSim/blob/master/AirLib/include/vehicles/multirotor/api/MultirotorRpcLibAdapators.hpp#L42-L77), as in `MSGPACK_DEFINE_MAP`, constructor and `to()`? You can see the definitions of other structs for different data types such as `Vector3r`, `float`, etc. (definations are there in `RpcLibAdaptors.cpp` also if you want to have a look)

Add an accessor method in [`MultiRotorApiBase.hpp`](https://github.com/microsoft/AirSim/blob/master/AirLib/include/vehicles/multirotor/api/MultirotorApiBase.hpp#L41), implement in [`SimpleFightApi.hpp`](https://github.com/microsoft/AirSim/blob/master/AirLib/include/vehicles/multirotor/firmwares/simple_flight/SimpleFlightApi.hpp#L140) call in [`getMultiRotorState()`](https://github.com/microsoft/AirSim/blob/master/AirLib/include/vehicles/multirotor/api/MultirotorApiBase.hpp#L125)
Might need to add some dummy functions in other firmwares for compilation

You'll need to add the field in Python API [`types.py`](https://github.com/microsoft/AirSim/blob/master/PythonClient/airsim/types.py#L315) also to receive it

This should _probably_ be enough, there's a lot of stuff since it supports multiple firmwares, etc. Though it could possibly be reduced
Might have missed something, getting late here. But should give some places to look at

rajat2004, 2020-06-19:
Also, make sure that you're writing your script in `PythonClient/multirotor`, `car` or CV folder, and before doing `import airsim`, use `import setup_path`. This will use the local airsim Python package in the repo instead of the one installed using `pip`, otherwise the extra field added won't actually be used
`setup_path` is used in many if not most of the example scripts in those folders

nicocarb, 2020-06-19:
Great, this worked. The last step I was missing was the field in `types.py`

closed 2020-06-19 by nicocarb