microsoft/AirSim #2785: AirSim ROS wrapper with Python API

pengzhi1998, 2020-06-21:
Hi everyone, is there a convenient way to use ROS wrapper with Python for Airsim?

pengzhi1998, 2020-06-21:
I found this [code](https://github.com/microsoft/AirSim/blob/master/PythonClient/multirotor/kinect_publisher.py#L134) a good example. It imports rospy and AirSim at the same time. However, if I didn't remember wrong, ROS uses Python 2 while AirSim should use Python 3. Is there a way to fix that? And will Melodic be applicable?

EdwinMeriaux, 2020-06-21:
Wait are you just trying to use ROS and Airsim together? Did you do the Airsim ROS wrapper set up?

pengzhi1998, 2020-06-21:
Thank you for your reply!
Sorry, I'm new on this topic. But I found that the AirSim ROS wrapper can only use C++ API. But I need to use Python API with Ubuntu 18.04. Do you know how to deal with that?

EdwinMeriaux, 2020-06-21:
Yeah you can just use python with that wrapper (at least from what I have used so far it works):

https://microsoft.github.io/AirSim/airsim_ros_pkgs/
follow these steps

thejeshk, 2020-06-23:
Hi @pengzhi1998 @EdwinMeriaux 

I have posted my query here -> https://github.com/microsoft/AirSim/issues/2796#issue-643951496

Can you confirm if this is valid for Car ?

EdwinMeriaux, 2020-06-24:
@thejeshk Sorry if what is valid for Car? Could you please clarify?

thejeshk, 2020-06-25:
Hi @EdwinMeriaux 

Thank you for your reply. 

I am using Airsim car not Multirotor. 

My question is: Is there a ROS wrapper for Car? 

I could only find the tutorials for drones https://microsoft.github.io/AirSim/airsim_tutorial_pkgs/

EdwinMeriaux, 2020-06-26:
I have never used the a car in airsim. But I would assume that you could write a similar one for the car.
What do you want to do the the car and ros?

madratman, 2020-07-29:
PR https://github.com/microsoft/AirSim/pull/2743 added this. This was also answered here https://github.com/microsoft/AirSim/issues/2796#issuecomment-650674502

closed 2020-07-29 by madratman