microsoft/AirSim #2788: SimpleFlight and state estimation

JacopoPan, 2020-06-22:
Is [this](https://github.com/microsoft/AirSim/blob/master/docs/simple_flight.md#state-estimation)

> In **current release we are using ground truth from simulator for our state estimation**. We plan to add complimentary filter based state estimation for angular velocity and orientation using 2 sensors (gyroscope, accelerometer) in near future. **In more longer term, we plan to integrate another library to do velocity and position estimation using 4 sensors (gyroscope, accelerometer, magnetometer and barometer) using EKF**. If you have experience this area than we encourage you to engage with us and contribute!

still the case (I would believe so, by looking at [AirSimSimpleFlightEstimator.hpp](https://github.com/microsoft/AirSim/blob/master/AirLib/include/vehicles/multirotor/firmwares/simple_flight/AirSimSimpleFlightEstimator.hpp)) and then something that might be worthwhile contributing to?

madratman, 2020-07-29:
Yes, that is still the case. And is definitely something worth contributing to via a PR.  
If you need help / feedback on the implementation, let's chat on this thread.

subedisuman, 2021-11-12:
Hi, I am implementing an EKF-based state estimator in AirSim simple_flight as described in the issue above (using 4 sensors) as part of my Master's Thesis at TUM and IABG. My implementation is still in a very early phase, however, I would much appreciate your feedback. Here is the branch to the forked repository: https://github.com/subedisuman/AirSim/tree/ekf. I am looking forward to your comments and hope my developments will be beneficial for the AirSim community.

jonyMarino, 2022-01-07:
Hi @subedisuman! That's great to hear. I will take a look at your fork when having time. I would love to see this becoming a PR.

subedisuman, 2022-04-20:
# Update
In the following [branch](https://github.com/subedisuman/AirSim/tree/ekf "branch"), I have implemented an Extended Kalman Filter (EKF) based state estimator as a part of my Master Thesis. It uses the following sensor measurements: IMU, GPS, Barometer, and Magnetometer to estimate the following states: local (NED) x-y-z positions, x-y-z velocities, attitudes, and IMU and barometer sensor biases. I have documented the changes I made below. There is also a demo to try things out and see results in plots :). I have tested it using Linux (Ubuntu 20.04.4 LTS) build of AirSim and UE 4.25. I would say it is content-wise complete and would appreciate your comments and remarks to help make improvements and run tests and validations. I would also like to know the next steps towards a pull request. Thanks :).

## Work so far
### Files Added
- [AirSimSimpleEkf.hpp](https://github.com/subedisuman/AirSim/blob/ekf/AirLib/include/vehicles/multirotor/firmwares/simple_flight/AirSimSimpleEkf.hpp "AirSimSimpleEkf.hpp"): contains the main Extended Kalman Filter (EKF) logic that uses IMU measurements to predict the states and GPS, barometer, and magnetometer measurements to update the states.
- [AirSimSimpleEkfBase.hpp](https://github.com/subedisuman/AirSim/blob/ekf/AirLib/include/vehicles/multirotor/firmwares/simple_flight/AirSimSimpleEkfBase.hpp "AirSimSimpleEkfBase.hpp"): contains common getters and setters for the AirSimSimpleEkf class.
- [AirSimSimpleEkfModel.hpp](https://github.com/subedisuman/AirSim/blob/ekf/AirLib/include/vehicles/multirotor/firmwares/simple_flight/AirSimSimpleEkfModel.hpp "AirSimSimpleEkfModel.hpp"): contains mathematical models of the aircraft kinematics and measurement models for sensors.
- [AirSimSimpleEkfParams.hpp](https://github.com/subedisuman/AirSim/blob/ekf/AirLib/include/vehicles/multirotor/firmwares/simple_flight/AirSimSimpleEkfParams.hpp "AirSimSimpleEkfParams.hpp"): contains reading and storage of configurations and parameters required by the EKF.
- [IEkf.hpp](https://github.com/subedisuman/AirSim/blob/ekf/AirLib/include/vehicles/multirotor/firmwares/simple_flight/firmware/interfaces/IEkf.hpp "IEkf.hpp"): contains the pure abstract class interface of EKF.
- [EkfStructs.hpp](https://github.com/subedisuman/AirSim/blob/ekf/AirLib/include/common/EkfStructs.hpp "EkfStructs.hpp"): contains declaration of structs used by EKF in `msr/airlib` namespace.

### Files Modified
#### Main Modifications
- [AirSimSimpleFlightBoard.hpp](https://github.com/subedisuman/AirSim/blob/ekf/AirLib/include/vehicles/multirotor/firmwares/simple_flight/AirSimSimpleFlightBoard.hpp "AirSimSimpleFlightBoard.hpp"): sensors pointers are introduced as state variables that are intitialized during construction of the flight board. Additional methods are introduced that gets sensor outputs from the sensor pointers. These methods are used by EKF to get sensor measurements.
- [AirSImSimpleFlightEstimator.hpp](https://github.com/subedisuman/AirSim/blob/ekf/AirLib/include/vehicles/multirotor/firmwares/simple_flight/AirSimSimpleFlightEstimator.hpp "AirSImSimpleFlightEstimator.hpp"): added getters for EKF estimated states that are distributed from here to other AirSim components. Can toggle between EKF estimated states and ground truth states.
- [AirSimSimpleFlightCommon.hpp](https://github.com/subedisuman/AirSim/blob/ekf/AirLib/include/vehicles/multirotor/firmwares/simple_flight/AirSimSimpleFlightCommon.hpp "AirSimSimpleFlightCommon.hpp"): added namespace conversion of structs used by EKF from `msr/airlib` to `simple_flight` and vice versa.
- [SimpleFlightApi.hpp](https://github.com/subedisuman/AirSim/blob/ekf/AirLib/include/vehicles/multirotor/firmwares/simple_flight/SimpleFlightApi.hpp "SimpleFlightApi.hpp"): added construction of AirSimSimpleEkf and public getters of EKF states.
- [Firmware.hpp](https://github.com/subedisuman/AirSim/blob/ekf/AirLib/include/vehicles/multirotor/firmwares/simple_flight/firmware/Firmware.hpp "Firmware.hpp"): added reset and update of EKF.
- [OffboardApi.hpp](https://github.com/subedisuman/AirSim/blob/ekf/AirLib/include/vehicles/multirotor/firmwares/simple_flight/firmware/OffboardApi.hpp "OffboardApi.hpp"): added minor commlink message to know if EKF states are used throughout (if EKF is enabled?).
- [CommonStructs.hpp](https://github.com/subedisuman/AirSim/blob/ekf/AirLib/include/vehicles/multirotor/firmwares/simple_flight/firmware/interfaces/CommonStructs.hpp "CommonStructs.hpp"): added definition of structs used by EKF in `simple_flight` namespace.
- [IBoardSensors.hpp](https://github.com/subedisuman/AirSim/blob/ekf/AirLib/include/vehicles/multirotor/firmwares/simple_flight/firmware/interfaces/IBoardSensors.hpp "IBoardSensors.hpp"): added interfaces for the sensor board that reads sensor outputs. These are used by EKF.
- [IStateEstimator.hpp](https://github.com/subedisuman/AirSim/blob/ekf/AirLib/include/vehicles/multirotor/firmwares/simple_flight/firmware/interfaces/IStateEstimator.hpp "IStateEstimator.hpp"): added interfaces for getters of EKF states.

#### Additional Modifications
- [AirSimSettings.hpp](https://github.com/subedisuman/AirSim/blob/ekf/AirLib/include/common/AirSimSettings.hpp "AirSimSettings.hpp"): added EKF Settings, attached it to vehicle settings, from where EKF Params reads the EKF specific settings added for this implementation.
[... cut ...]
       "D:\a\AirSim\AirSim\AirSim.sln" (default target) (1) ->
       "D:\a\AirSim\AirSim\AirLibUnitTests\AirLibUnitTests.vcxproj" (default target) (11) ->
       (ClCompile target) -> 
         D:\a\AirSim\AirSim\AirLibUnitTests\main.cpp : fatal error C1128: number of sections exceeded object file format limit: compile with /bigobj [D:\a\AirSim\AirSim\AirLibUnitTests\AirLibUnitTests.vcxproj]

    0 Warning(s)
    1 Error(s)
```
    
There is already a solution mentioned, but I am not sure about this.

jonyMarino, 2022-04-26:
Thank you very much @subedisuman, for the detailed update. You are doing an excellent job! Keep it up!

subedisuman, 2022-08-03:
Hi @jonyMarino. Thank you for your kind remarks. Regarding the recent announcement to close the support for this repository, I wonder how it affects this potential pull request? I would love to get your feedback regarding this issue and the new strategy. The current implementation can make use of the AirSim imu, gps, barometer, and magnetometer sensor signals to compute and use estimated states based on configurable EKF parameters. Ongoing work-in-progress is to test the implementation. If possible, I would also love to get your feedbacks regarding the functionality, the implementation, the procedures to test it, and the next steps regarding the potential pull request. Thank you.