microsoft/AirSim #2789: ROS node "pid_position_controller_simple_node" and launch file "airsim_with_simple_PD_position_controller.launch"

JacopoPan, 2020-06-22:
I was looking at the code in [`pd_position_controller_simple.cpp`](https://github.com/microsoft/AirSim/blob/master/ros/src/airsim_ros_pkgs/src/pd_position_controller_simple.cpp), is it correct to assume that:
1. On lines 334 and 338, the 4 occurrences of `vel_cmd_.twist.linear.z` should be replaced with `vel_cmd_.twist.angular.z` 
2. To get rid of `[ERROR] [1234567890.1234567890]: [PIDPositionController] Waiting for odometry!`, one should replace line 64 
```
airsim_odom_sub_ = nh_.subscribe("/airsim_node/odom_local_ned", 50, &PIDPositionController::airsim_odom_cb, this);
```
With something like 
```
airsim_odom_sub_ = nh_.subscribe("/airsim_node/VEHICLE_NAME/odom_local_ned", 50, &PIDPositionController::airsim_odom_cb, this);
```

3. Finally, [airsim_with_simple_PD_position_controller.launch](https://github.com/microsoft/AirSim/blob/master/ros/src/airsim_ros_pkgs/launch/airsim_with_simple_PD_position_controller.launch) throws an error which might be due to the double inclusion of `static_transforms.launch` (in `airsim_node.launch` and `airsim_with_simple_PD_position_controller.launch` itself)

Zartris, 2021-02-19:
So this is some time ago this was proposed, and I'm still getting the odom error.
I have not followed these yet, but I was wondering if this proposed fix is true and why they haven't implemented it yet if so.

rajat2004, 2021-02-21:
I guess it's more due to the lack of any PR. I've made some commits on this branch - https://github.com/rajat2004/AirSim/tree/ros-fixes, if possible, could you test it out?
There are currently some problems with my OpenCV install, that's why unable to test

JacopoPan, 2021-02-21:
True @rajat2004, I never made a PR for the `airsim_odom_sub_` subscription because, at the time I created this issue, I wasn't sure of how to retrieve `vehicle_name` from within `PIDPositionController::initialize_ros()`

leoborgnino, 2021-03-03:
Hi guys! I'm Leandro. I fixed this with rosparams. In airsim_ros_wrapper.cpp , after "parsing" settings.json, I set rosparam "/vehicle_name" to the parsed vehicle name.  In pd_position_controller I get this rosparam and I use it to the subscription to airsim_odom_sub_.

This is a different approach than @rajat2004 's one but I think it could be better because the user doesn't have to change the .launch file.  Theoretically, if you clone the repo, you should be able to launch everything without changes.

I think the solution could be better if the parameter sync can be improved (it works but is not so elegant). In addition, it doesn't consider the existence of multi-vehicles, but that is easy to change.

The branch:
https://github.com/leoborgnino/AirSim/tree/fix_ros_name_topics

If someone wants to test it or to implement a more elegant rosparam sync would be great. Also, if someone can see the solution (it's only 3 lines) and give their opinion!

abdo-ameen, 2021-06-15:
@JacopoPan I implemented the correction above:
`airsim_odom_sub_ = nh_.subscribe("/airsim_node/drone_1/odom_local_ned", 50, &PIDPositionController::airsim_odom_cb, this);`
where `drone_1` is my vehicle name, then I ran `catkin_make` in the root folder `/AirSim/ros` and finally ran the launch file `roslaunch airsim_ros_pkgs position_controller_simple.launch`, but I'm still getting the same error:
`[ERROR] [1623703365.547240000]: [PIDPositionController] Waiting for odometry!`

moreover, I cloned @leoborgnino 's branch and then ran the launch file:
`roslaunch airsim_ros_pkgs position_controller_simple.launch`
and I'm still getting the same error:
`[ERROR] [1623703337.583211200]: [PIDPositionController] Waiting for odometry!`

I cloned @rajat2004 's branch too and did the same thing, but no change.

Any idea about the cause of this error?
I'm using Windows 10 with WSL2 for ROS Noetic, AirSim 1.5 with Unreal 4.26.

abdo-ameen, 2021-06-16:
Thank you @leoborgnino for your response. After the `airsim_node` get initialized, it publishes into `/airsim_node/drone_1/odom_local_ned` correctly. This is the echo output

[image]

and this is the info of the topic when both `airsim_node` and 'pid_position_node` are launched
[image]

and of course the resulting error
[image]

leoborgnino, 2021-06-16:
> Thank you @leoborgnino for your response. After the `airsim_node` get initialized, it publishes into `/airsim_node/drone_1/odom_local_ned` correctly. This is the echo output
> 
> [image]
> 
> and this is the info of the topic when both `airsim_node` and 'pid_position_node` are launched
> [image]
> 
> and of course the resulting error
> [image]
> 
> this is the msg info in case it can help, because in `pd_position_controller_simple.cpp` there is a comment that says "todo check if odometry is too old!!". I'm not sure what does it mean
> [image]

Great! It seems to be ok. I have a question, the error "waiting for odometry" appears all the time, right ? Because in my env it starts printing "waiting for odometry" a few times but then it stops, and that is normal.

abdo-ameen, 2021-06-17:
That is right! It's working fine, just throwing the error a couple of times at the beginning! Absolutely my bad!
[image]

Thank you so much @leoborgnino for taking the time and responding :innocent: :blossom:

zimmy87, 2021-07-14:
Hi @leoborgnino, I've been investigating an issue related to this one, #3805. Your changes in https://github.com/leoborgnino/AirSim/tree/fix_ros_name_topics are useful for fixing #3805 as well. Would you be amenable to publishing a PR from your branch so we can merge your changes into master?

zimmy87, 2022-01-21:
The fix for this issue has been merged as part of #3880, so I am closing this issue.

closed 2022-01-21 by zimmy87