microsoft/AirSim #2792: Physics of the Multirotor

nmuchaxo, 2020-06-22:
Hello, let me start by saying I'm New to UE and AirSim. 

I've been playing with the Blocks world for some days and I created a radial impulse which will affect the objects around it. A ball and the vehicle are in the radius of the impulse when I press play, however my vehicle is not affected by the impulse and the ball is. 
Anyone has any idea of why this is happening? 
TIA

nicocarb, 2020-06-22:
From my experience, AirSim object do not respond to Unreal objects unless specified in the AirSim code. For example, in FastPhysicsEngine there is a function called getKinematicsOnCollision that details how the vehicle should react if it collides with an object in Unreal. If you want to implement this, I would imagine you would have to do something similarly with impulses though I am not sure. Keeping thread open as I don't have any immediate solution

nmuchaxo, 2020-06-23:
Hello nicocarb, thanks for your answer. 
Yes, I'm looking at that right now, but I think this is only for collisions, unfortunately it doesn't cover impulses as you said, altough an impulse is like a collision with an invisible object... 
Has anyone done this with impulses? Do you think it's like change one thing here and there or do I need to create a new function like 'getKinematicsOnImpulse' and rewrite a bunch of stuff?...

nicocarb, 2020-06-23:
I imagine it's a larger change than just one thing. I would first go about seeing if you can detect an impulse through AirSim and then create a function like 'getKinematicsOnImpulse' if you can