microsoft/AirSim #2795: Fire detection

Al-Kaff, 2020-06-23:
Hello,
I am working on designing a scenario containing several fire shapes and sizes for fire detection purposes.
I have created the fires using UE4 particle systems, however, when I start the simulation with AirSim, I can see the fire in the RGB image, but I cannot detect it neither in segmentation nor infrared.

It is possible to solve this issue?

Regards

ivannson, 2020-07-01:
I think that _might_ be because there is no collision box around fire, as it's obviously not a physical object. I think you might have to label fire manually with bounding boxes.

rptamin, 2021-05-14:
The infrared image is based upon the segmented image from RGB. The segmented image is created by assigning a value to each static mesh in Unreal. Considering that fire is a particle effect in Unreal, is there a way to add fire as a static mesh?

jbericat, 2022-01-22:
Hello Abdulla, I've been working on an AirSim project quite similiar to the one you're describing (fire detection and stuff), and I also run into the same problem you're facing. Actually, what I had to do is to add random static mesh objects under the fire VFX's, and then aim for the staticmesh objects instead of the fire (using the capture_ir_segment.py script). Then, I collected both the RGB and the segmented image around that static mesh object. After that I overlapped the segmented image over the RGB one (converted to grayscale to simulate a FLIR camera image). Finally I ended up achieving an effect such as in this image:

[image]

As you may notice this workaround won't fix the problem, but in my case was more than enough to achieve what I wanted to do (a proof of concept system where a NN model can distingish among different kind of simulated wildfires). 

I Hope it helps, but since I actually have little info about your use case, it might not be sufficient for you. Are you doing this for a bachelor's degree assignment, as I were? or are you doing it just for fun?

Good luck!

andrewjong, 2022-01-27:
Looks great @jbericat ! What are your thoughts on how to reduce the smoke appearing in the final output? After all, a thermal camera is meant to help see through smoke, right? In these [example videos from FLIR](https://www.flir.com/discover/government-defense/combating-california-wildfires-from-above-with-the-flir-star-safire-380-hdc/), it seems at most only the base of the smoke plume shows hot on the thermal camera. Perhaps decreasing the alpha on the RGB portion would work?

jbericat, 2022-01-29:
Hey @andrewjong thank you for your comment! :D

Right now, I've just finished the project related where I had to implement the FLIR simulator -that is, my  B.S. Computer Science's Final Degree Project, which you can find here if you're interested ->  https://github.com/UOC-Assignments/uoc.tfg.jbericat. However, since I'm committed to keep researching on this area and I already got another project on the oven, then I decided to re-think the whole FLIR-sim thing from scratch just using a different approach (one which I still have to meditate a lot about, though). Anyway, for the new project I still have to decide between using the Unreal Engine platform and NVIDIA's Omniverse -The new project involves creating a real area digital twin and I'm DYING to try the Omniverse thing! I also have to perform a deeper research on ways to simulate wildfire evolution and how factors such as the terrain orthography and climatology influence it during it's lifetime, but I'm getting a bit of help on that already.

So, returning back to your question: 

> Looks great @jbericat ! What are your thoughts on how to reduce the smoke appearing in the final output? After all, a thermal camera is meant to help see through smoke, right? 

If I had to go back in time to do things differently, then I'd have added the fire VFX 's just without the smoke emitters -all the images I used to train the NN model were FLIR simulated, so there was no point on adding smoke. Anyway, that was just a proof-of-concept / toy-app aimed to demonstrate how to train a model using synthetic data -I just did choose this scenario because my willing to keep studying this matter on the future.

Sorry if I couldn't provide a more sophisticate answer... ^^' 

Cheers!