microsoft/AirSim #2796: ROS wrapper for car

thejeshk, 2020-06-23:
Hello, 

Can anyone please tell me how to use ROS Wrapper for car. I want to use Publish images package -> object detection (YOLO) node and move_base "navigation package" for navigation

But for now I want to use object detection node https://github.com/leggedrobotics/darknet_ros with my Car camera images as input 

Thank you 
TK

thejeshk, 2020-06-23:
I am using 

 - Ubuntu 18.04
 - Airsim 1.3
 - ROS Melodic 

Vehicle used: 
[image]

My settings file: 

{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "ClockType": "",
  "ClockSpeed": 1,
  "LocalHostIp": "127.0.0.1",
  "RecordUIVisible": true,
  "LogMessagesVisible": true,
  "ViewMode": "",
  "RpcEnabled": true,
  "EngineSound": true,
  "SimMode": "Car",
  "ViewMode": "SpringArmChase",
  "ClockSpeed": 1.0,
  "PawnPaths": {
    "DefaultCar": {"PawnBP": "Class'/AirSim/VehicleAdv/X950R/X950R_Pawn.X950R_Pawn_C'"}
  },
  "Vehicles": {
    "car_1": {
      "VehicleType": "PhysXCar",
      "DefaultVehicleState": "",
      "AutoCreate": true,
      "PawnPath": "",
      "EnableCollisionPassthrogh": false,
      "EnableCollisions": true,
      "RC": {
        "RemoteControlID": -1
      },
      "Sensors": {
        "Imu" : {
          "SensorType": 2,
          "Enabled": true
        },
        "GPS":{
          "SensorType": 3,
          "Enabled": true
        },
        "LidarCustom": {
          "SensorType": 6,
          "Enabled": true,
          "NumberOfChannels": 1,
          "PointsPerSecond": 10800,
          "X": 0,
          "Y": 0,
          "Z": -1,
          "DrawDebugPoints": true
        }
      },
      "Cameras": {
        "front_left_custom": {
          "CaptureSettings": [
            {
              "PublishToRos": 1,
              "ImageType": 0,
              "Width": 672,
              "Height": 376,
              "FOV_Degrees": 90,
              "TargetGamma": 1.5
            }, 
            {
              "PublishToRos": 1,
              "ImageType": 1,
              "Width": 672,
              "Height": 376,
              "FOV_Degrees": 90,
              "TargetGamma": 1.5
            }
          ],
          "X": 2, "Y": -1, "Z": -1,
          "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
        },
        "front_right_custom": {
          "CaptureSettings": [
            {
              "PublishToRos": 1,
              "ImageType": 0,
              "Width": 672,
              "Height": 376,
              "FOV_Degrees": 90,
              "TargetGamma": 1.5
            }
          ],
          "X": 2, "Y": 1, "Z": -1,
          "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
        }
      },
      "X": 2, "Y": 0, "Z": 0,
      "Pitch": 0, "Roll": 0, "Yaw": 0
    }
  },
  "SubWindows": [
    {"WindowID": 1, "ImageType": 0, "CameraName": "front_left_custom", "Visible": true},
    {"WindowID": 2, "ImageType": 0, "CameraName": "front_right_custom", "Visible": true}
  ]
}

GimpelZhang, 2020-06-28:
Here is a PR which provides the car ros node for AirSim：#2743

It is not merged yet, but I think it will be soon.

thejeshk, 2020-06-29:
Thank you @GimpelZhang

closed 2020-06-29 by thejeshk