microsoft/AirSim #2800: error: msgpackrpc.error.RPCError

gulbulut, 2020-06-25:
Hi,
Using Ubuntu 18.04 LTS and latest AirSim version. I get the following error when I run my application that captures image with sitl and ros2 node. unreal engine is not installed, instead airsim works on a project.  Should I install the Unreal engine? Will airsim not work from a file produced with Unreal Engine?

> msgpackrpc.error.RPCError: rpclib: function 'enableApiControl' (called with 2 arg(s)) threw an exception.

gulbulut, 2020-06-25:
Fixed. I updated about the Airsim python library. However, I cannot say this is the definitive solution.

closed 2020-06-25 by gulbulut

thejeshk, 2020-11-23:
Hi @gulbulut 

Does your Ros2 node work with the latest AirSim version? 

BEst Regards, 
TK