microsoft/AirSim #2804: LiDAR as training data

tiger-bug, 2020-06-26:
Good afternoon, 

This may already be an implementation so feel free to close this if this is the case.  I am attempting to use AirSim to collect labeled LiDAR training data.  I know that you can collect LiDAR already and have a python script to save to disk, but is there a way to collect the label from the points as well?  I've only been using this for a couple of days.  I saw that [Issue 1878](https://github.com/microsoft/AirSim/issues/1878) had the same question.  Was this ever resolved?  My OS is Ubuntu 18.04, AirSim version is 1.3.1, Python is 3.8 (conda environment), and Unreal version is 4.24.  Thanks!

rajat2004, 2020-06-26:
Maybe [`simGetLidarSegmentation`](https://github.com/microsoft/AirSim/blob/master/PythonClient/airsim/client.py#L539)?

rajat2004, 2020-06-29:
FYI, a separate API for segmentation data being produced by Lidar didn't exactly make sense to me, since the data is already being produced by Lidar, and another call can give mismatched data. I've opened #2810 which combines the segmentation data into `LidarData` struct

ivannson, 2020-07-01:
I did exactly that with `simGetLidarSegmentation`. It provides the labels for all of the points as a separate array. The order is the same, i.e. point in position 23 corresponds to label in position 23

closed 2020-08-17 by jonyMarino