microsoft/AirSim #2805: Gimbaled Stereo Camera on Drone

Omar-Elmofty, 2020-06-27:
I am trying to simulate a gimballed stereo camera that is attached to the drone. I realize that there is a gimbal that can be setup for individual monocular cameras, however I am wondering if there is a simple way to setup the gimbal for a stereo pair? 

If I have to develop a custom solution for this problem, I will have to be able to modify the position and orientation of each of the cameras forming the stereo pair. I know you can change  the camera's orientation using the python API call simSetCameraOrientation, however how can you modify the camera's position? I looked at issue #1049 but it's not clear to me how it was resolved!

rajat2004, 2020-06-27:
There is a PR https://github.com/microsoft/AirSim/pull/2710 which upgrades the existing `simSetCameraOrientation` to `simSetCameraPose` and allows setting of position & orientation, that might be what you're looking for

Omar-Elmofty, 2020-06-27:
This works! Thanks for your help!

closed 2020-06-27 by Omar-Elmofty

rajat2004, 2020-06-28:
Glad to hear, do comment on the PR if any more documentation is required or any examples should be added