microsoft/AirSim #2806: Distance Detection Right in Front of Drone

EdwinMeriaux, 2020-06-27:
Hello, 

So I would like to be able to detect the distance from just one point in front of the drone it see if there is anything immediately in front of it.

I know I cannot just use the distance sensor (since they can only for now detect distance from the drone to floor) But is there anyone way I can shift that to just in front of the drone. Or can I just use the Lidar to only detect 1 point in front of the drone?

rajat2004, 2020-06-27:
I'm working on adding a setting for drawing debug point in Distance Sensor (https://github.com/microsoft/AirSim/pull/2807), and during testing I'm seeing different behaviour actually.
When the distance sensor is enabled, I'm getting some 22m distance, and the debug point is also coming in front of the drone rather than below it. When the drone hovers, I'm still getting the same distance.
My code for drawing the point could be wrong, but the values should mean that I'm measuring the front distance rather than the bottom. Or am I doing something wrong in testing?

EdwinMeriaux, 2020-06-28:
Wait sorry I am confused. You say you get some 22m distance. Is it detecting a object 22m in front that is actually there or is it a false positive?

rajat2004, 2020-06-28:
I'm using the Blocks env, and yes, there's an object in front of it at 22m
[image]

After a good night's sleep, I can see that my code right now is wrong, the position of the Debug Point remains the same even when the vehicle moves since I'm using the `relative_pose` to the vehicle, the vehicle pose needs to be added to it to get actual moving pose of the sensor in NED frame.
So it could actually be that after applying the fix, the point could even come at the bottom, but the distance measurements only make sense if it's measuring the front of the vehicle.
Also, see https://github.com/microsoft/AirSim/pull/791#issuecomment-364314524, in which it's mentioned that the sensor is actually measuring the front of the vehicle

rajat2004, 2020-06-28:
@EdwinMeriaux I've updated the documentation in #2807, and have tested it to be working alongwith modifying the settings to face the distance sensor downwards, and the values also make sense. It would be great if you could give it a try

Downward facing sensor -
- On ground => `'distance': 0.38477620482444763,`
- Hovering => `'distance': 2.021547794342041,`

EdwinMeriaux, 2020-06-29:
Awesome! Thank you so much! I really would like to try it.
I just have one question though. Is the set up the same as the regular old distance sensor?

rajat2004, 2020-06-30:
I'm not able to understand exactly what's the confusion, usage of Distance Sensor is the same as before. Just need to enable it in the vehicle's sensors. You can see the updated doc in the PR, the orientation, etc of the sensor can be specified in the settings, and there's an example of how to orient the sensor downwards so that it measures altitude. 
Do comment on the PR if more info regarding something is required

EdwinMeriaux, 2020-07-03:
(sorry for the delay in responding) So I know how to set up the distance sensor in normal case (measuring the distance to floor) but what do I need to do to use the new forward facing distance sensor?

rajat2004, 2020-07-03:
That's the current confusion, from my testing, the Distance sensor measures in front by default, and specific settings are needed to point it towards the ground. Could you explain as to how are you testing that it's measuring towards the ground by default?
I've updated the sensor doc at https://github.com/microsoft/AirSim/pull/2807/files#diff-1077f6c1304b766013eb174bb5e5321d

EdwinMeriaux, 2020-07-03:
Oh. I am so sorry. I completely misunderstood that. I just double checked and yes the distance sensor works in the forward direction I was under the impression that the data was from the downward facing direction. I apologize again for the confusion on my part.

rajat2004, 2020-07-03:
Ahh great, thanks for testing! This actually got me thinking a bit, and then I realised that this will probably need some fixes to get it working with PX4

EdwinMeriaux, 2020-07-03:
I have never used a PX4 in Airsim. What kind of fixes would you think you need?

rajat2004, 2020-07-03:
I've pushed a commit which I think should work, but needs testing. I have used PX4 earlier, but not with AirSim. Will try to get it setup and test in the next few days

PR #2807 (2020-07-24): Distance sensor Upgrades
1. Add setting for drawing debug point with Distance Sensor
2. Expose Distance sensor settings, also move sensor above Car vehicle by 1 meter by default 
3. Fix `DistanceSensorData` struct in `types.py` (Should be float, not Quaternion)
4. Update docs
5. Use Custom sensor orientation in Mavlink Distance Sensor message (Needs testing with PX4)
6. ROS wrapper change
7. Example script for Distance sensor with 2 cars

Default sensor-
[image]

Downward-facing (altitude measurement)-
[image]

2 Cars with distance sensor
[image]

Closes #2723, closes #2806
Related issues - #2 [... cut]
closed 2020-07-24 by madratman