microsoft/AirSim #2811: Deep Q Learning

johan606303, 2020-06-29:
Hi. 

I made a code for obstacle avoidance, but there is an error that I can't understand the origin.
At a random iteration (example : 400, 1208,...) VS2019 interrupt the training, and returns me the following error.
[image]

There is the full code. As I'm new with RL, it's not a complexe code. And where there are variables inside tim.sleep(x), that are divided by 300, it's because I've accelerated the time of UE4.

Thak you for help.

`
import numpy as np
import airsim
import time
import math
import tensorflow as tf
import keras
from airsim.utils import to_eularian_angles
from airsim.utils import to_quaternion
from keras.layers import Conv2D,Dense
from keras.layers import Activation
from keras.layers import MaxPool2D
from keras.layers import Dropout
from keras.layers import Input
import keras.backend as K
from keras.models import load_model
from keras import Input
from keras.layers import Flatten
from keras.activations import softmax,elu,relu
from keras.optimizers import Adam
from keras.optimizers import adam
from keras.models import Sequential
from keras.optimizers import Adam, RMSprop
from keras.models import Model
#tf.compat.v1.disable_eager_execution()
import random

from collections import deque

client=airsim.MultirotorClient()
z=-5
memory_size=10000000
#pos_0=client.getMultirotorState().kinematics_estimated.position

#state_space=[84, 84]
#action_size=3

def OurModel(state_size,action_space):
    
    X_input=Input(state_size,name='Input')
    X=Conv2D(filters=32,kernel_size=(8,8),strides=(4,4),padding='valid',activation='relu')(X_input)
    X=MaxPool2D(pool_size=(2,2))(X)
    X=Conv2D(filters=64,kernel_size=(4,4),strides=(2,2),padding='valid',activation='relu')(X)
    X=MaxPool2D(pool_size=(2,2))(X)
    X=Conv2D(filters=64,kernel_size=(1,1),strides=(1,1),padding='valid',activation='relu')(X)
    
    X=Flatten()(X)

    X=Dense(525,activation='relu')(X)
    X=Dropout(0.3)(X)
    X=Dense(300,activation='relu')(X)
    X_output=Dense(action_space,activation='softmax')(X)
    
    model=Model(inputs = X_input, outputs = X_output)
    model.compile(loss="mse", optimizer=RMSprop(lr=0.0005, rho=0.95, epsilon=0.01), metrics=["accuracy"])
    model.summary()

    return model

class MemoryClass():
[... 208 lines cut ...]
                if done==True:
                    total_reward.append(sum(episode_rewards))
                    self.memory.add([next_state,action,state,reward,done])
                    self.step=self.max_step
                    self.reset()
                    print("episode: {}, epsilon: {:.5}, total reward :{}".format(episode, self.epsilon,total_reward[-1]))
                    self.save("airsim-dqn.h5")
                else:
                    state=next_state
                    self.memory.add([next_state,action,state,reward,done])
                if len(self.memory.buffer)>64:
                    next_states_mb, actions_mb, states_mb, rewards_mb,dones_mb=self.memory.replay()
                    target = self.model1.predict(states_mb)
                    target_next = self.model1.predict(next_states_mb)
                    for k in range(len(dones_mb)):
                        if dones_mb[k]==True:
                            target[k][actions_mb[k]] = rewards_mb[k]
                        elif dones_mb[k]==False:
                            target[k][actions_mb[k]] = rewards_mb[k] + self.gamma * (np.amax(target_next[k]))
                    self.model1.fit(x=states_mb,y=target,batch_size=self.batch_size)

agent=Agent()
agent.train()

`

rajat2004, 2020-06-30:
UE can sometimes fail to render the image, and in that case, the image would be empty. You'll have to handle this case, maybe call the API again if the image is empty?
Note that right now, Airsim fills the array with a single 0 since when the code was written, `rpclib` had a bug with handling empty structs - https://github.com/rpclib/rpclib/issues/152, so you'll have to check if the length is 1, then handle it accordingly
Since the RPC problem has been fixed, I think the workaround in AirSim can be removed now

A discussion could be had whether an empty image should be handled by Airsim internally or by the user

johan606303, 2020-06-30:
@rajat2004 An another question if possible. Is there a way, in AirSim, to reset position of 1 drone only. For example if I try to use DQL or RL for several multirators ?

rajat2004, 2020-07-01:
Hmm, the current `reset()` API doesn't take in Vehicle name as an arg, but it should be possible to use it on a per-vehicle basis - https://github.com/microsoft/AirSim/blob/master/AirLib/src/api/RpcLibServerBase.cpp#L183. But not sure as to how it'll affect the physics
As a shortcut for now, you could try using `client.simSetVehiclePose(airsim.Pose(airsim.Vector3r(0, 0, 0), airsim.to_quaternion(0, 0, 0)), True, vehicle_name='')`

closed 2020-07-12 by johan606303