microsoft/AirSim #2813: Surface normals image reference frame

marcociara379, 2020-06-30:
Hello everyone,
I have a question about the **surface normals**  image:
what is the reference frame the normals are in? I would expect NED (world frame), although from `SceneTexture:WorldNormal` node I would expect ENU (as per UE convention). 

Does someone know better?

Thank you

LSBOSS, 2020-06-30:
I plugged together a small test scene (with 9 axis aligned cubes and a camera looking down the positive x-axis) as I was about to report an issue with surface normals having a small bias applied when routed through AirSim. But I can take the very same image to answer your question:

[image]

When observing the color on top of the cube (127,127,255) and route them through the usual normal unpacking (color value / 255 * 2.0 - 1.0) you get (0,0,1) as a normal. That's a hint for ENU. If surface normals would have been in NED space the normal's z-value would be negative.

marcociara379, 2020-06-30:
Thanks a lot for your hints @LSBOSS.
I came to the same conclusion playing a little with the UE editor. 
Nonetheless, the values that I get instead of 127 are 187, that I do not get where it comes from (I am still using UE4.18 and AirSim 1.2 due to some internal development of mine). 
Is this issue what you mean when you refer to "small bias"?

LSBOSS, 2020-06-30:
> Thanks a lot for your hints @LSBOSS.
> I came to the same conclusion playing a little with the UE editor.
> Nonetheless, the values that I get instead of 127 are 187, that I do not get where it comes [...]

Yes, that's the one. Thanks for commenting on my opened issue https://github.com/microsoft/AirSim/issues/2815 :D

madratman, 2020-07-29:
https://github.com/microsoft/AirSim/pull/2845 closes this

closed 2020-07-29 by madratman