microsoft/AirSim #2817: Question on ros wrapper

pengzhi1998, 2020-06-30:
Hi everyone, sorry to bother you again.
I installed ROS wrapper as guided [here](https://github.com/microsoft/AirSim/tree/master/ros/src/airsim_ros_pkgs). However, when I ran `roslaunch airsim_ros_pkgs airsim_node.launch`, the command line only suggests: "Waiting for connection" and a lot of "X" appears (I was assuming a simulator might be opened, but nothing happened except the "waiting"). Is there anything wrong with my operation? Or should I open anything else?
Any guidance will be greatly appreciated!

drbswa, 2020-06-30:
You have to make sure that the simulation is running first before launching. If you're using UE, make sure you hit play first, then go launch.

pengzhi1998, 2020-07-04:
Got it, thank you!

closed 2020-07-04 by pengzhi1998